micropython/micropython #2971: RFC: defining machine.Timer

dpgeorge, 2017-03-21:
This issue is to discuss the API of the machine.Timer class.  There are currently 3 different version of Timer which don't have much in common:

esp8266:
```python
# id must be -1, period is in milliseconds, mode is PERIODIC or ONE_SHOT
Timer(id, *, period, mode, callback) # constructor
Timer.init(*, period, mode, callback) # reinitialise all parameters
Timer.deinit() # stop the timer
```

cc3200/wipy:
```python
# mode is PERIODIC, ONE_SHOT or PWM
Timer(id, mode, *, width=16) # constructor
Timer.init(mode, *, width=16) # reinitialise
Timer.deinit() # stop

# each timer has one or more channels
# each timer channel is associated with a specific IO pin (eg for PWM)
Timer.channel(*, freq, period, polarity, duty_cycle) # create channel assoc with TImer
TimerChannel.freq([freq]) # get/set
TimerChannel.period([period]) # get/set
TimerChannel.duty_cycle([duty]) # get/set
TimerChannel.irq(...) # get or set up the IRQ callback
```

stmhal/pyboard (actually pyb.Timer):
```python
# freq is in Hz (can be float <1), prescaler/period are raw values, mode is UP/DOWN/CENTER
Timer(id, *, freq, prescaler, period, mode, div, callback, deadtime) # constructor
Timer.init(*, freq, prescaler, period, mode, div, callback, deadtime) # reinit
Timer.counter([value]) # get/set the raw timer counter
Timer.source_freq() # get source frequency that clocks this timer
Timer.freq([value]) # get/set
Timer.prescaler([value]) # get/set
Timer.period([value]) # get/set
Timer.callback(callable) # get/set callback (pass None to disable)

# each timer has multiple chanels (usually 4)
# pin is the pin to associate with the channel (eg for PWM)
Timer.channel(mode, *, callback, pin, pulse_width, pulse_width_percent, compare, polarity) # create
TimerChannel.pulse_width([value]) # get/set
TimerChannel.pulse_width_percent([value]) # get/set
TimerChannel.capture([value]) # get/set
TimerChannel.compare([value]) # get/set
TimerChannel.callback(callable) # get/set callback
```

To compare current usage, for a simple periodic timer at 10Hz one would do:
```python
# esp8266
t = Timer(-1)
t.init(period=100, mode=Timer.PERIODIC, callback=lambda t:print(t))

# stmhal/pyboard
Timer(1, freq=10, callback=lambda t:print(t))
```

For a one-shot after 2 seconds:
```python
# esp8266
t = Timer(-1)
t.init(period=2000, mode=Timer.ONE_SHOT, callback=lambda t:print(t))

# stmhal/pyboard
def cb(t):
    print(t)
    t.deinit() # or t.callback(None)
Timer(1, freq=0.5, callback=cb)
```

At the very least I'd suggest that all ports have the same interface to create periodic and one-shot timers, as well as the same way to configure the timer irq/callback.  Configuration of the irq should follow the samy signature as machine.Pin.irq(); see also #2297.

I also feel that a timer should have start() and stop() methods, so that one can create an initialise a timer at one point in the code, then start it precisely when needed.  This requires definining that the first callback happens one period after start() is called.

crespum, 2017-07-12:
@dpgeorge has there been any progress on this topic? I'm willing to help because the lack of a generic interface is pretty confusing.

dpgeorge, 2017-07-13:
@crespum no progress, this ticket got a bit lost... feel free to make suggestions for the interface.

crespum, 2017-07-24:
@dpgeorge I've started by adding `Timer.period([value]) # get/set` for the ESP8266 to understand how to contribute and also because I needed it. Shall I open a PR for [this small improvement](https://github.com/crespum/micropython/tree/esp8266/timer)?

crespum, 2017-07-28:
According to @dpgeorge's proposal, [I've created](https://github.com/crespum/micropython/tree/esp8266/timer) the following interface for the ESP8266:

```python
Timer(id) # constructor
Timer.init(period, mode=PERIODIC) # reinitialise all parameters
Timer.deinit() # disarm timer
Timer.start() # start timer
Timer.stop() # stop timer
Timer.period([period]) # get the period/set new period
Timer.irq(handle) # configures timer callback
```

An example on how to use it would be:
```python
t = Timer(-1)
t.init(period=100, mode=Timer.ONE_SHOT)
t.irq(lambda t:print(t))
t.start()
```

I somehow feel that `Timer.init()` should be completely replaced by `Timer.start()`. The same for `deinit` and `stop`. Any suggestion?

stinos, 2017-07-29:
> I somehow feel that Timer.init() should be completely replaced by Timer.start()

In my opinion you have too much methods, and some of them might never get used, but if I had to get rid of some it wouldn't be by merging those 2, at least they make sense: it allows you to have one component setup a Timer, than pass it to another which can just use start/stop without having to know aby more details. Such decoupling is handy.
I would get rid of deinit() (if you have stop(), why do you need deinit()? - unless I'm misunderstanding what stop() does) and period() (you have init() already - also what happens exactly when you change a Timer's period when it is running?) and irq() can be merged in init() as well (what's the use of a Timer without a callback anyway?). And allow calling init() from the constructor by repeating all arguments there: thinking of how I usually use timers and see them used, it seems that setting them up is in > 90% of cases a one time thing, and timer instance which aren't set up are not useful at all; meaning in nearly all cases you will always construct then immediately call init(). So I'd have the code as easy to use as possible by expressing the most used case:

```
t = Timer(-1, period=100, mode=Timer.PERIODIC, callback=myTimerIRQ)
t.start()
t.stop()
```

peterhinch, 2017-07-29:
> pass it to another which can just use start/stop ... Such decoupling is handy.

Agreed.

crespum, 2017-07-31:
I pretty much agree with you. I added `Timer.irq()` to follow `Pin.irq()`'s signature, but I agree it doesn't make much sense having a timer without callback.

Regarding to `Timer.period([period])`, it's useful not only for modifying the timer's period while keeping the same callback function and mode, but also for retrieving the current period.

Would you agree with this?
```python
Timer(id, period=val, mode=PERIODIC, irq=handler) # constructor
Timer.init(period=val, mode=PERIODIC, irq=handler) # reinitialise all parameters
Timer.start() # start timer
Timer.stop() # stop timer
Timer.period([period]) # get the period/set new period
```

nickovs, 2018-06-21:
It seems that this discussion went sort of cold but it would be good to get some resolution on it, since the current model makes it very hard to write portable code.

There are currently a few problems with writing portable code that needs accurate timers. The function signature for `init()` is highly variable, the units are variable, obscure and undocumented and resolution is poor on platforms other than stm32 even when the hardware and RTOS can do a great deal better (such as on the ESP32). 

I'm about to take a look at fixing the ESP32 timer code so that it's easier to use the available resolution. For the time being I will implement this by allowing for a keyword `freq` that takes a float of the frequency in Hz and use it in preference to the `period` when present, like the stm32 port currently does. Personally I'd like to see a `freq` parameter, explicitly stated as having a unit of Hz, being in the standard signature.

On a more general note, there is currently a idiom of getting values on machine objects using `val = object.attr()` and setting them using `object.attr(val)`, which is not very _Pythonic_. Now that accessors have been [enabled by default](https://github.com/micropython/micropython/commit/93150a0d40a6f29907d53bba8a4a44f1c5fb2449) how would people feel about switching to `val = object.attr` and `object.attr = val`?

nickovs, 2018-06-21:
It would also be helpful for the timer API to have a `period_hz` attribute that would let you find out the unit for the period setting. For things like the ESP8266 this could be stuck to 1000 and reject being changed but for things like the stm32 it could be changeable and replace the need for `prescaler`.

crespum, 2018-06-22:
A while ago I did some progress on #3253, but the lack of a common interface meant that making this changes was really difficult. I hope you find that PR useful, if not feel free to close it.

peterhinch, 2018-06-22:
> On a more general note, there is currently a idiom of getting values on machine objects using val = object.attr() and setting them using object.attr(val)

This was discussed at length vs. the use of properties. Use of methods is advocated for performance reasons because the lookup for properties is much slower. There is also a view that assigning to properties should not trigger substantial side effects (like, say, initiating transmission of data across a slow interface). Using a function or method to trigger such behaviour is more orthodox.

I think similar reasoning will apply to descriptors. There is a difference in philosophy between MicroPython and CircuitPython in this respect; personally I hope MicroPython retains its performance focus.

nickovs, 2018-06-22:
@peterhinch Thanks for that background information. I can understand the desire to not have substantial side-effects and there is certainly an overhead associated with having properties or descriptors.

nickovs, 2018-06-22:
@crespum Thanks for your PR. I think that until there is a solid consensus as to what the common API should look like I'm not going to make an changes that are not backward-compatible.

After a little digging it seems that the ESP8266 does have a microsecond accuracy timer and the timer on the ESP32 can also do better. For the time being I will just add to these two ports the ability to optionally pass a floating point number as the period in milliseconds and support passing a float for the frequency in Hz instead, and keep everything else the same, even though it irks me that we won't have `freq == (1/period)` like physics says we should!

dpgeorge, 2018-06-29:
> the units are variable, obscure and undocumented

Yes, and I think this should be fixed by using "natural" units (eg `freq` should be in Hz), and/or explicitly specifying the units as a suffix (eg `period_ms`).  This is how `time.sleep_ms()` and friends are written and I think works well (and note that without a suffix, `time.sleep()` is in natural units of seconds).

> Personally I'd like to see a freq parameter, explicitly stated as having a unit of Hz, being in the standard signature.

I agree.

> and keep everything else the same, even though it irks me that we won't have freq == (1/period) like physics says we should!

I guess the motivation for having `period` in ms is because in seconds you'd almost always need to use decimal values, and floats should be optional in the API (not all ports have them, they require heap memory allocation, they are slower).  But then probably it's better to go for microseconds to give more accuracy.

But since the different ports assign different meanings to `period` already, maybe we just leave it alone for now and instead introduce `period_ms` and/or `period_us`?  Then potentially deprecate `period` and repurpose it later on as seconds.

dpgeorge, 2018-06-29:
> On a more general note, there is currently a idiom of getting values on machine objects using val = object.attr() and setting them using object.attr(val) ...
> how would people feel about switching to val = object.attr and object.attr = val?

>This was discussed at length vs. the use of properties. Use of methods is advocated for performance reasons because the lookup for properties is much slower. There is also a view that assigning to properties should not trigger substantial side effects

I don't know if it's worth the effort to dig this old issue back up, and then end up with the same conclusion.  Performance of attributes is an issue, but maybe an efficient solution could be found if attributes are really "the right thing".  Currently how it's implemented there are two ways to implement attributes at the C level:
1. add them as functions to the `locals_dict` of the class wrapped in property accessors (like `staticmethod` and `classmethod` work); pro: uses a hash table to look names up, so is efficient; con: lots of code bloat needed to wrap each attribute in a property accessor
2. make the C type implement the general `->attr()` method and then it handles all gets/sets/deletes itself; pro: very efficient in code size because one function can do all gets/sets of attributes; con: slow to lookup entries because it must search everything with if-else (for minimal code size).

Point 2 could be improved in efficiency by searching `locals_dict` before dispatching to `->attr()`.  It would need some looking in to.

Regardless of implementation efficiency/code-size, the two main arguments against attributes are: 1) get/set of an attribute should be a simple operation, but some config options do a lot of work behind the scenes; 2) it's sometimes hard to draw the line between what is a simple attribute and what should be a method call, so making everything a method call eliminates the need to make this decision for each new item.

dpgeorge, 2018-07-05:
The discussions above, and in #3901, #3922 and #3923 have resulting in the following ideas for setting the timer period/freq, for which I'd like to make a full proposal to move forward with:
1. Timer constructor (and init method) has `freq`,`period` and `period_hz` keyword-only arguments (among others).
2. If `freq` is given it sets the frequency of the timer in Hz.  Ints must be supported.  Floats may be supported.
3. If `period` is given (but not `period_hz`) it sets the period in milliseconds.  It is an int.
4. If `period` and `period_hz` are given it sets the period in seconds as `period / period_hz`.  `period_hz` is an int.

Points 3 and 4 mean that `period_hz` has a default value of 1000.  Ie the default units of `period` are milliseconds.

So the signature of the constructor would look something like:
```python
Timer(id, *, freq=None, period=None, period_hz=1000, ...)
```
The advantage of this is that you can set the frequency or period in whatever the natural units are for the problem.  And `period/period_hz` gives a possible range of period between 10^9 seconds and 1 nanosecond using small integers.

I'd propose that all ports (where possible) be updated to support this signature, as well as the docs.

nickovs, 2018-07-05:
> sets the period in seconds as `period / period_hz`

This seems to be the [rational](https://en.wikipedia.org/wiki/Rational_number) way to do things 😄

nickovs, 2018-07-05:
> I'd propose that all ports (where possible) be updated to support this signature, as well as the docs.

For clarity, would you expect to leave, deprecate or remove other keyword arguments from the existing implementations? The existing STM32 code currently has explicit `prescaler` and `div` values, the use of which requires knowledge of the hardware. While these allow you to make the most of the quirky STM32 hardware they also make code that uses them non-portable.

nickovs, 2018-07-05:
One other thing that we might want to consider in the API is to explicitly support a `Timer` object attribute calls something like `context` that just holds a reference to some object that might be used by the callbacks and pass it as a parameter to the callback if it's not None. Nearly every callback hander that I write starts with some `global blah` because I want to update something from inside the handler. Formalising the API to support a context pointer in the init method would cost one `mp_obj_t` of memory per allocated timer but would make it easier to write generic handler functions and if we pass it as a parameter then it saves a dictionary lookup inside the time-critical code. If we do this for timer callbacks we should also do it for IRQ handlers.

dhylands, 2018-07-05:
I find using explicit prescaler/div values to be most useful when setting up the channels as capture channels. In these scenarios you often use a very fast timer in the MHz range. Even setting up PWM for servo control, I typically use a 1 MHz timebase so I can get microsecond resolution on the pulse width.

So even though its not portable, I think its still worth leaving in.

nickovs, 2018-07-05:
@dhylands In the case that you have do you specifically need access to the hardware prescaler and div, or do you just need access to faster, higher resolution timers? If you need direct access then we should keep that option but if your ends can be satisfied using a more abstract and portable interface then we might be able to clean things up.

rolandvs, 2018-07-05:
Lets keep the prescaler and div as they are close(r) to the hardware and needed in certain cases as Dave has highlighted.

dhylands, 2018-07-05:
> @dhylands In the case that you have do you specifically need access to the hardware prescaler and div, or do you just need access to faster, higher resolution timers? If you need direct access then [...]

I think its really about when you have multiple registers interacting. Just picking a frequency isn't sufficient. For example, with input capture you have the underlying frequency of the timer, and the register which determines the capture time. There may be lots of combinations that give a correct underlying frequency but which don't give the appropriate range/resolution on the other registers.

Neon22, 2018-07-05:
A while ago I wrote a helper for the Papillio FPGA clock. This was a complex clock. The helper let you define the base frequency and then ask for suggestions on which internal clocks, and which prescalers set to what values, would get you to the desired clock output.
https://raw.githubusercontent.com/Neon22/papilio-clock/master/papilio-clock_samples-1.png

In an FPGA this is quite an annoying problem. Its a suprset of the problem @dhylands  mentions.
Would it be useful to construct a helper tool for this prescaler problem to help calculate the proper values of the div and prescaler could be set ?
Of course it might need options for each set of micropython supported hardware...
- https://github.com/Neon22/papilio-clock

dpgeorge, 2018-07-06:
> For clarity, would you expect to leave, deprecate or remove other keyword arguments from the existing implementations? The existing STM32 code currently has explicit prescaler and div values...

For stm32 code in particular, the existing class is really `pyb.Timer` which is distinct (at least in theory) from `machine.Timer`.  It's because of legacy/history that `pyb.Timer` exists and `machine.Timer` is trying to be a portable replacement for the former (same with other classes in these modules).  But `pyb.Timer` will stick around for some time for backwards-compatibility.

I do want to provide the user with detailed control of the stm32 timer hardware (as well as other peripherals) and it's still unclear how `machine.Timer` could do this in a portable way.

> Nearly every callback hander that I write starts with some global blah because I want to update something from inside the handler.

Did you try using closures to capture state?

> Formalising the API to support a context pointer in the init method
> ...
> If we do this for timer callbacks we should also do it for IRQ handlers.

This would need a separate discussion.  `machine.Pin` already has it's irq/callback behaviour defined (but could be extended).  And see this PR for an implementation of generic callback mechanism on stm32: https://github.com/micropython/micropython/pull/3854

dpgeorge, 2018-07-06:
> Is there a case for `period_exponent` where -9 is ns -3 is ms and so on? ... Surely we only need powers of 10 here.

That's a possibility... but one argument for `period_hz` is that you can tune the value to the frequency of the external event/device/etc that you use (eg 1500Hz) and then `period` is expressed in these units.  But an exponent would definitely allow a much larger range.

nickovs, 2018-07-06:
> > Is there a case for period_exponent where -9 is ns -3 is ms and so on? ... Surely we only need powers of 10 here.

> That's a possibility... but one argument for period_hz is that you can tune the value to the frequency of the external event/device/etc that you use (eg 1500Hz) and then period is expressed in these units. 

There are also cases where, if the user does know what the underlying hardware clock frequencies are, being able to specify the denominator as a factor of the base frequency leads to better accuracy.

> But an exponent would definitely allow a much larger range.

Given that `period_hz` allows for a range between one nanosecond and 317 year using only small ints it's not clear to me that there are use cases that need a much larger range, let alone hardware that will support it. Furthermore I would posit that by the time that we have access to hardware that can get close to using the low end of that range the standard int representation will allow for periods between one fifth of an attosecond and five billion trillion years. If someone can come up with a use case that needs more than that then I'm happy to listen but I confess I'm somewhat skeptical.

nickovs, 2018-07-06:
>> Nearly every callback hander that I write starts with some global blah because I want to update something from inside the handler.

> Did you try using closures to capture state?

Doh! No, I didn't. In fact for the purposes of the code in hand bound methods work even better.

OK, ignore that suggestion!

dpgeorge, 2018-07-11:
Ok, so, the pros of `period_hz` are that the name is short to type (compared to something like `period_denominator`), it tells you the units, and allows for quite a large range and a high level of accuracy when specifying a period.  The cons are that it is somewhat confusing at first look (this can be addressed with good docs and examples), requires extra typing compared to something like `period_us` (could add `period_us` later as well if really needed), and doesn't have as large a range as specifying an exponent (could use floating point in `freq` instead, or allow a bignum for `period_hz`).

Given that there are lots of good points about it, I'd say we go for `period_hz`.

nickovs, 2018-07-20:
I thought it would be good to record here that in light of the discussion of #3923 the new timer code on the stm32, esp8266 and esp32 ports uses `tick_hz` for the denominator of the `period` value rather than `period_hz`.

nickovs, 2018-07-20:
I have been writing some documentation for the new `Timer.init()` method signature and in the light of doing this and the examples that I have been writing I would like to propose an API optimisation.

On all of stm32, esp8266 and esp32 it is now possible to specify either a frequency in Hz using the `freq` argument or a time period in milliseconds using `period` or in seconds using `period` and `tick_hz`. Looking at all the code I've written and all of the examples that I was writing for the docs, I noticed that whenever I wanted a periodic timer I was specifying frequency and whenever I needed a one-shot timer I was specifying a time duration. I think that this is down a pretty fundamental difference in the way one conceptualises periodic events vs one-shot events.

So, how would people feel if we made the `Timer.init()` function not always require the `mode` argument but instead made `mode` default to `PERIODIC` if a frequency is given and default to `ONE_SHOT` if a time duration is given?

I appreciate that there are arguments against the behaviour being implicit rather than explicit but on the other hand the (a) the behaviour only changes when you change which (mutually exclusive) keyword arguments you use, (b) you can still choose to specify if you wish and (c) it is going to guess the correct behaviour the vast majority of the time while making the code smaller.

dpgeorge, 2018-07-27:
> made the `Timer.init()` function not always require the mode argument but instead made mode default to PERIODIC if a frequency is given and default to ONE_SHOT if a time duration is given?

I see how this could shorten code, and make typing at the REPL a lot easier if you want the default.  But I'd be cautious not to overload the behaviour of a constructor/method too much.  Imagine someone reading code and seeing `Timer(1, freq=10)`, and then in another part of the code `Timer(1, period=100)`.  I think it would be expected that these two pieces of code give the same result (ie they are both periodic).

And I think this change would be backwards incompatible: the implementations of esp8266 and esp32 have `mode` as a keyword-only argument which defaults to periodic in all cases.  Also stm32 defaults to periodic (but it's `mode` argument has different meaning to esp8266/esp32).  The current docs for machine.Timer describe the cc3200 port only and they need to be updated (it's unfortunate there is so much discrepancy here, but the whole aim of this discussion is to clean it up).