micropython/micropython #2973: RFC extmod/modframebuf.c Support for blitting from 1-bit to n-bit colour maps

peterhinch, 2017-03-21:
We now have 1-bit horizontal maps. One use for these is for rendering bitmaps such as font glyphs by blitting from a framebuf containing the glyph to the destination framebuf. This works - in a geometry agnostic manner - on monochrome displays where the destination uses 1-bit colour mapping.

Where the destination uses n-bit colour maps the resultant colour is unexpected. The general solution is to introduce the concept of palettes defining a mapping between differing colour spaces. This is not 'micro'.

A 'micro' solution might be to add an optional colour arg to the blit method which, when the source framebuf has a 1-bit mapping, causes the blit to be rendered in the supplied colour. This would enable fast font rendering in arbitrary colours and would be easily implemented.

Or is there a better approach?

dpgeorge, 2017-03-22:
I was thinking the same thing, to add a "palette" argument to blit.  Blitting lower bit-depth images to larger bit-depth buffers seems to be the most common thing to do, so the palette can actually be quite micro: for 1-bit source it's just 2 entries, and for 4-bit source (GS4_HMSB format) it's just 16 entries.  The latter is only a 32 byte palette when blitting to a 16-bit RGB framebuf.

deshipu, 2017-03-23:
I'd love to have a stab at implementing this.
Another thing that I noticed is that even with rather big palettes, there are going to be very regular -- usually the values are just scaled up/down to cover the larger/smaller color space. So maybe it would make sense to do it with a callback function, that would compute the new color on the fly without having to waste memory for all the color entries? We can have some common functions (scale, threshold, palette, invert, etc.) provided for convenience, but also let the user provide their own. This would also be useful for transforming images with the same color depth.
But I'm afraid that I'm over-generalizing here and that calling a python callback for every pixel would make it as slow as just looping over the pixels in Python, defeating the purpose of the blit function (even if we memoized it).

deshipu, 2017-07-24:
I'm sorry for the long delay, but I finally made the first draft of the code, here: https://github.com/micropython/micropython/pull/3238

As you can see, it's extremely simple right now. You just pass an array of ints defining the mapping.

peterhinch, 2019-06-17:
Given that this seems to have died can I propose a "micro" solution?

This caters for the common case where the source has a 1-bit mapping and the destination has a greater depth: for example where glyphs are to be rendered to colour displays.

In this solution, two additional optional args to **blit** provide foreground (FG) and background (BG) colours in a format compatible with the destination framebuf. If FG is provided and the source pixel is nonzero the pixel is rendered in FG. If BG is provided and the source pixel is zero the pixel is rendered in BG.

This should require minimal code, have little impact on performance and radically improve the speed of a common use-case. Rendering glyphs to colour displays using Python is excruciatingly slow compared to mono (where **blit** can be used).

I'd be happy to submit a PR if there's any interest in this.

peterhinch, 2021-08-25:
Closing this as now fixed by https://github.com/micropython/micropython/pull/7682.

closed 2021-08-25 by peterhinch