micropython/micropython #2976: Soft reboot is unreliable if there are timers and interrupts at play

jcea, 2017-03-22:
Frequently when I do a soft reboot (control+d or via software "sys.exit()") of my ESP8266, if there are timers at play, the reboot if started but the ESP hangs. For Instance:

```
PYB: soft reboot
#6 ets_task(40100164, 3, 3fff8398, 4)
Traceback (most recent call last):
  File "<stdin>", line 72, in 
(the device hangs)
...
(watchdog reset after some time)
```

I think that the first step of the soft reboot should be to disable timers and ISR's.

dpgeorge, 2017-03-23:
> I think that the first step of the soft reboot should be to disable timers and ISR's.

Yes, that will certainly help, and it would be good to make the esp8266 able to soft-reboot more reliably than it currently does.  But it may not be possible to do it in a completely clean way because of the wifi stack, which is difficult to clean up correctly.

pfalcon, 2017-04-05:
See also https://github.com/micropython/micropython/issues/1896

dpgeorge, 2020-06-09:
esp8266 soft-reset has improved a lot over the years, especially due to b2fa1b50edaea43007d692d24391c6f3aa4d15ca.  So I'll close this issue for now.

closed 2020-06-09 by dpgeorge