micropython/micropython #2977: "self" corrupting during dict-lookup inside list-comprehension with super() call

waveform80, 2017-03-25:
I've come across a rather strange issue when writing some modules for Micropython running on an ESP32. However, I've managed to reproduce the issue on the Unix port so I don't *think* this is anything specific to that port.

The issue is that the "self" object seems to be corrupted when performing a dict-lookup inside a list comprehension involving a super() call. The following code should demonstrate the issue (at least it does for me on both the Unix port and the ESP32 port):

```python
class A():
    def __init__(self, start):
        self._start = start

    def get_events(self):
        return list(range(self._start, self._start + 5))

class B(A):
    def __init__(self, start):
        super().__init__(start)
        self._lookup = {i: i * 2 for i in range(10)}

    def get_events(self):
        return [self._lookup[i] for i in super().get_events()]

a = A(1)
print(a.get_events())

b = B(1)
print(b.get_events())
```

The expected output is:

```
[1, 2, 3, 4, 5]
[2, 4, 6, 8, 10]
```

However, on the Unix port I get:

```
[1, 2, 3, 4, 5]
Traceback (most recent call last):
  File "super_issue.py", line 20, in <module>
  File "super_issue.py", line 14, in get_events
  File "super_issue.py", line 6, in get_events
AttributeError: '' object has no attribute '_start'
```

(on the ESP32 port I just get a crash and reboot)

I realize this test-case seems a little convoluted, but I've tried to pare it down to the bare minimum and if I remove the `super()` call, or the dict-lookup, or the list-comprehension from the test everything works happily, so I'm guessing it's some interaction of all three of these.

Just in case it helps narrow things down: in my original code the "A" class was implemented in C, while the "B" sub-class was in Python (I originally thought that might be the root of the problem, but evidently I was wrong, having reproduced the issue in pure Python). However, while messing around with this in C I did note that the "self" pointer is wrong during the failed `b.get_events()` super-class call (which isn't quite as easy to see in the pure Python test-case).

Final note for anyone running into this: it's pretty easy to work-around the issue. Just construct the list with a for-loop instead of a list-comprehension and everything works nicely.

waveform80, 2017-03-25:
Additional: I had a cursory glance at the `super()` implementation in `objtype.c`. I don't pretend to understand most of it yet, but the comment in `super_make_new` "0 arguments are turned into 2 in the compiler" caught my attention. I wondered if the compiler is getting this transformation wrong, so I tried modifying the critical line in the test case to:

```python
return [self._lookup[i] for i in super(B, self).get_events()]
```

This appears to fix things, so my guess is that this is an issue in the micropython compiler.

pfalcon, 2017-03-25:
@waveform80 : Thanks for the report. Did you try to produce a shorter testecase (e.g. one class, other things removed, etc.).

waveform80, 2017-03-25:
Yes: I tried without the `super()` call (one class) which failed to reproduce, without the list-comprehension (also failed to reproduce) and without the dict-lookup (also failed to reproduce). Whatever the issue is, it's something which requires all of these in combination. As noted above, it also requires it to be a bare `super()` call, as `super(B, self)` also fails to reproduce the issue.

My apologies that it's a complex reproduction case, but it really is the simplest I've managed to make it so far.

waveform80, 2017-03-25:
Additional information: digging into the compiler, it looks like `compile_trailer_paren_helper` in `py/compile.c` is responsible for transforming `super()` into `super(some-class, some-instance)`. The special chunk at the top handles this and contains the comment 'look for first argument to function (assumes it's "self")'. Presumably, the compiler's looking at the enclosing scope here and trying to find the first argument to the function-call, which would indeed normally be "self". But I wonder what happens if we're within, say, a lambda in that function, or a closure in that function, or in the case in question a list-comp within that function?

I'll see if I can produce something a little more minimal armed with this (though it'll have to involve two classes just for the `super()` call)...

pfalcon, 2017-03-25:
> though it'll have to involve two classes just for the super() call)...

Formally, not necessarily, you can use super() in a class which doesn't inherit from anything, because even in that case the class still inherits from `object`. But you'd be limited in what you call by super() of course (likely just constructor). Anyway, if you tried to play with the code to minimize it, and the above is what you come to, that's ok, it should be good enough for @dpgeorge to look into it when he finds time. Thanks.

waveform80, 2017-03-25:
Oh dear ... I'm beginning to get the feeling this might actually be a more general Python issue / feature. My original test case (above) works on Python 3.5.2 (Ubuntu 16.04's standard build), and fails on Micropython, hence my assumption that it was an issue with Micropython. However, while attempting to construct a more minimal test-case, based upon the idea that it's the compiler improperly expanding `super()` to `super(foo, bar)` in certain complex scopes, I came up with the following:

```python
class A():
    def __init__(self):
        self._start = 1

    def foo(self):
        return self._start

class B(A):
    def foo(self):
        return [super().foo() for i in range(5)]

B().foo()
```

This fails under Micropython ... and Python 3. I confess, I hadn't realized just how magic Python 3's bare `super()` call was (as it turns out [very](https://mail.python.org/pipermail/python-3000/2007-June/008301.html) [very](http://lucumr.pocoo.org/2010/1/7/pros-and-cons-about-python-3/) [magic](http://stackoverflow.com/a/19609168/1459019)).

It also turns out, after a bit more googling, this issue [is known in Python 3](http://stackoverflow.com/questions/31895302/python3s-super-and-comprehensions-typeerror). This doesn't explain why the original test case above works in Python 3 (by all rights, it shouldn't given in the information in that SO answer), but I'm reasonably satisfied at this point that *if* there's an issue in Micropython here, it's that Micropython doesn't simply raise `TypeError` (like Python 3) but tries to execute code with an invalid "self" reference.

I'll leave this open for now so someone more qualified can judge the right course of action here. My impression at the moment is that Micropython should be raising `TypeError` in the face of these test cases but it seems even Python 3's implementation differs in its response to them!

peterhinch, 2017-03-26:
Interesting. Both code samples work if the call to ``super()`` in the comprehension is replaced with ``super(B, self)``. In the comprehension's scope, this seems to be the correct syntax. So the issues with the bare ``super()`` are the crash on the ESP8266 and the error messages produced.

dpgeorge, 2017-03-27:
Thanks for the report, the bug (the first example code) should be fixed by 40c1272e5516b8d847118c06c7d222410075fa09.

Some background: a list/set/dict comprehension has the outermost iterator evaluated within the outer scope, but the rest is evaluated within the comprehension's own scope.  And super() will only work in a function scope (not a comprehension scope).  Therefore, with
```python
[foo(x) for x in iterator()]
```
the `iterator()` expression is evaluated in the outer scope and `foo(x)` is evaluated in the comprehension's scope.

So you can use super() in the iterator (assuming the outer scope is a function within a class) but not in the expression that's evaluated for the list elements.  That's why the first example above works in CPython but the second example doesn't.

uPy also has the same behaviour as CPy in this respect, but there was a bug that only surfaced because of other factors: the variable "self" was being closed-over because it was referenced in the main expression (foo(x) above), as well as being needed by super() in the iterator expression.  But the compiler didn't take into account the fact that self could be closed over and the super() expression just loaded the local directly, when it should have dereferenced the closure cell.  This bug would surface in other example where self was closed over, not necessarily using a comprehension.

This is now fixed.

dpgeorge, 2017-05-06:
All the code snippets here now run the same on uPy as CPy, so closing the issue.

closed 2017-05-06 by dpgeorge

waveform80, 2017-05-06:
Fantastic - many thanks for the quick fix!