micropython/micropython #2982: Qstr generation does not kick in when changing header files

stinos, 2017-03-28:
As briefly discussed in #2974. I looked into this and found the cause plus possible solution, so this is a new issue for that. To reproduce, on unix:
- make clean
- in mpconfigport.h, delete the line with #define MICROPY_STACK_CHECK (1)
- make
- restore said line mpconfigport.h
- make -> MP_QSTR_stack_use is undeclared in modmicropython.c

The reason is that although in py/py.mk the target for $(HEADER_BUILD)/qstrdefs.generated.h has mpconfigport.h (and mpconfig.h and makeqstrdata.py etc) as a dependency, this is of no use since the sources including mpconfigport.h are not scanned again for qstrs: that is done in $(QSTR_DEFS_COLLECTED) and dependent targets but they don't kick in since they have no dependency on mpconfigport.h.

Now since basically all source files include mpconfigport.h/mpconfig.h/.. I think the solution is fairly simple: detect when mpconfigport.h (or one of those other files) change and if so do a complete preprocessor/qstr generation run again. Which is pretty close to what make clean/make would do, the only solution now, so it's no overkill and since it should only be a couple of extra lines in the makefile I think it's justified.

I don't have enough experience with makefiles to express this correctly so I'll leave the final solution to someone else. I did verify the principle works though, and it does. I used this sledgehammer to crack the nut:
```
-$(HEADER_BUILD)/qstr.split: $(HEADER_BUILD)/qstr.i.last
+$(HEADER_BUILD)/qstr.i.last.all: $(SRC_QSTR) $(PY_SRC)/makeqstrdata.py mpconfigport.h $(MPCONFIGPORT_MK) $(PY_SRC)/mpconfig.h | $(HEADER_BUILD)/mpversion.h
+       $(ECHO) "GEN $@"
+       $(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $^ >$(HEADER_BUILD)/qstr.i.last;
+
+$(HEADER_BUILD)/qstr.split: $(HEADER_BUILD)/qstr.i.last $(HEADER_BUILD)/qstr.i.last.all
 ```
It solves the problem but obviously is far from ideal.
In pseudo-code I assume an actual solution would be to adjust the qstr.i.last target like:
```
$(HEADER_BUILD)/qstr.i.last: $(SRC_QSTR) $(PY_SRC)/makeqstrdata.py mpconfigport.h  $(MPCONFIGPORT_MK) $(PY_SRC)/mpconfig.h | $(HEADER_BUILD)/mpversion.h
  if( $?.contains_any_of($(PY_SRC)/makeqstrdata.py mpconfigport.h  $(MPCONFIGPORT_MK) $(PY_SRC)/mpconfig.h))
    PP_SRC = $(SRC_QSTR)
  else
    PP_SRC = $?
  endif
  $(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $(PP_SRC) >$(HEADER_BUILD)/qstr.i.last;
```

pfalcon, 2017-04-05:
@dhylands : Ping.

dhylands, 2017-04-05:
The real issue seems to be dependencies.

If we break things down, we have some output from the pre-processor (let's call it .qstr_pp file). Currently the dependencies have foo.o depend on foo.c and all of the headers that it #includes. To make the dependencies work properly, what we really want is to have the foo.qstr_pp depend on the exact same things that the foo.o depends on. Then, since the qstrdefs.generated.h file depends on the .qstr_pp file, and the .o file depends on the qstrdefs.generated.h this will ensure that qstrdefs.generated.h is built before we compile.

We probably want to filter out qstrdefs.generated.h from the dependencies of the .qstr_pp file, so we don't have circular dependencies.

But the way that current makefile has cpp processing multiple source files gets really tricky to get the dependencies setup properly. I understand that we're doing this for performance. Anyways, I'll need to think on this some more to see if there is a way we can get around it.

pfalcon, 2017-04-05:
Well, I didn't try to go thru this in detail myself, but for me, the keywords are following:

> but they don't kick in since they have no dependency on mpconfigport.h.

So, maybe we just need to check why those things don't have automatic deps, or maybe just sprinkle explicit deps on it somewhere?

stinos, 2017-04-05:
> what we really want is to have the foo.qstr_pp depend on the exact same things that the foo.o depends on

That is the ideal case, but to figure out all headers foo.c (and hence foo.o) depends on you basically need the C compiler, no? Or can make do that?
So my idea was:
- I don't know how to figure out all fine-grained dependencies
- I'm not even sure it is needed since (AFAIK) mpconfig.h/mpconfigport.h and the likes are the only headers which determine what qstrs are available because configuration happens there though the preprocessor defininitions
- basically all C files depend on mpconfigport.h

Hence it seems logical to introduce just that dependency (i.e. mpconfigport changes => all sources must be scanned again) for the qstr generation?

dhylands, 2017-04-05:
> what we really want is to have the foo.qstr_pp depend on the exact same things that the foo.o depends on
> 
> That is the ideal case, but to figure out all headers foo.c (and hence foo.o) depends on you basically need the C compiler, no? Or can make do that?

As part of compiling, we already create dependency files, and those dependency files have all of the dependencies in them. In the compile_c macro, we already take the .d file produced by the compiler and massage it to produce a .P file. We could easily do further massaging to create an additional set of dependencies. The first time build (when the dependencies don't exist) is covered by the fact that the .qstr_pp file doesn't exist (in much the same way as compiling the .o file relies on the .o file not existing to cover the first build on a clean tree).

> So my idea was:
>
>     I don't know how to figure out all fine-grained dependencies
>     I'm not even sure it is needed since (AFAIK) mpconfig.h/mpconfigport.h and the likes are the only headers which [...]

I think that the issue is, it really depends on py/mpconfig.h, which in turn uses mpconfigport.h which may also in turn use mpconfigboard.h (depending on the port). There may also be other config variations which are port specific.

If all we wanted to do was to capture the mpconfig.h dependency, then we we could always create a dummy .c file that #included mpconfig.h (and any other headers deemed necessary) and create a special dependency file for just that one C file (using essentially the same technique used to create the current dependency files) and use that as part of the generated qstr dependency.

stinos, 2017-04-05:
> we already create dependency files, and those dependency files have all of the dependencies in them

Thanks for clearing that out, didn't know it happens like that.

> we we could always create a dummy .c ... and use that as part of the generated qstr dependency.

That sounds like it could do the trick.

dpgeorge, 2017-04-30:
I think github automagically closed this in error, because it's only fixed for msvc.

gojimmypi, 2017-06-19:
I see a similar problem with MICROPY_PY_SYS_STDIO_BUFFER in mpconfigport.h

[https://forum.micropython.org/viewtopic.php?f=16&t=3482](https://forum.micropython.org/viewtopic.php?f=16&t=3482)

pfalcon, 2017-08-23:
I'm getting this regularly, and really wishing that somebody fixed it. (Not yet pestered enough to go and dig this myself ;-) ).

dpgeorge, 2018-12-04:
This was addressed by 7ad04d17dabd503b0a79426166bcdc9e8ea4bf91

commit db9c2e31 (2017-04-30): msvc: Rebuild all qstrs when mpconfig headers are modified

Make qstr generation depend on modifications in mpconfigport.h, mpconfig.h
and makeqstrdata.py and if any of those change scan all source files for
qstrs again since they might have changed (for example typcially when
enabling new features in mpconfig.h).
This fixes #2982 for msvc builds.
closed 2018-12-04 by dpgeorge