micropython/micropython #2983: USB on Pyboard screws up buffers intermittently

kentindell, 2017-03-28:
After a lot of hunting (due to a soak test exploding) I traced a problem to the USB drivers. Sometimes (rarely) reading USB data gets back the data being transmitted. In other words, the buffers are somehow getting mangled.

I had a look through the USB handling code here: 

https://github.com/micropython/micropython/blob/master/stmhal/usbd_cdc_interface.c

It seems that all the buffer management (incrementing pointers etc.) is done without concurrency control and so there's a risk they are not atomic. The buffers would certainly go haywire then. This would also explain why it only happens intermittently.

Has anyone else noticed USB going a bit screwy? It would only be noticed after a long time (my soak tests see it after about 20 minutes).

dpgeorge, 2017-03-28:
Yes I have noticed this before, it's been around for a while but it is very rare and I don't have any reproducible cases.  I see it very occasionally when running the test suite for tests that output a lot of data.

Do you have a reproducible case?

> It seems that all the buffer management (incrementing pointers etc.) is done without concurrency control and so there's a risk they are not atomic

There is no threading so there isn't really any concurrency.  And any concurrency that may be there due to USB interrupts should be managed by the interrupt priorities, preventing reentrancy.  But for sure there is an issue and at this point everything is suspect.

kentindell, 2017-03-28:
I will see if I can synthesize a small example. But first I am also going to put ``raise_irq_pri(IRQ_PRI_OTG_FS)`` and  ``restore_irq_pri()`` calls around the buffer handling: incrementing an index/pointer with interrupts allowed is definitely a problem. Then if the problem goes away we probably know where it is..

kentindell, 2017-03-28:
I made the changes and my soak test now passes.

The changes were basically of this form:

    ...
    uint32_t basepri = raise_irq_pri(IRQ_PRI_OTG_FS);
    // Write data to device buffer
    UserTxBuffer[UserTxBufPtrIn] = buf[i];
    UserTxBufPtrIn = (UserTxBufPtrIn + 1) & (APP_TX_DATA_SIZE - 1);
    restore_irq_pri(basepri);
    ...

The above is in ``USBD_CDC_Tx``, but I also wrapped the increment code in ``USBD_CDC_Rx`` and ``USBD_CDC_TxAlways``.

I didn't guard the logic in the loops because I think any interleaving that results in the tests passing also results in the code subsequently using the buffer safely.

dpgeorge, 2017-03-29:
@kentindell thanks for the update on your tests.  I managed to make a test that reliably hits this bug (basically running the tests/stress/dict_copy.py test many times in a row and verifying the output).  The patch you provide above with raising IRQ doesn't fix it for me.  Instead I can fix it by increasing the timeout waiting for the endpoint to send the data, on line 265 of usbd_cdc_interface.c:
```
if (UserTxBufPtrWaitCount < 500) {
```
(timeout was 150, increased to 500).  Does that fix it for you, ie without raising the IRQ priority?

dpgeorge, 2017-03-29:
A timeout of 200 also seems to make the bug go away  Using 175 the bug is still there.

kentindell, 2017-03-29:
Just done some experiments. The timeout doesn't fix my bug, and it goes away again when I put the ISR locks back in.

I think there are two bugs here: one when the host PC going AWOL causes the USB drivers to stutter and the one I saw.

I should describe in more detail the symptoms of what I see going wrong. Basically I'm transferring real-time sensor data to the host PC using a transport protocol over serial. The protocol implementation is all driven by polling: no ISRs, so no concurrency issues. The issue I have is that sometimes, fairly rarely, calling ``USBD_CDC_Rx`` gives back data that had earlier been put into the _transmit_ buffer ``UserTxBuffer`` (via the ``USBD_CDC_TxAlways()`` call). Because my protocol is designed to recover and resync regardless of where it starts it erroneously picks out old transmitted frames and acts as if they were received (hence my soak tests exploding).

Those symptoms can happen if the receive pointer gets corrupted and ends up indexing into the transmit buffer. And that could happen if an ISR and a read call happened to both try to alter the same buffer pointers at the same time. This is of course a classic no-no with ISR programming, so these ISR lockouts over shared variable updates are always going to be necessary. When I put the lockouts in the bug goes away. So I think we can conclude that there's a classic shared variable concurrency bug here.

The critical sections are very short, so USB interrupts aren't locked out for any appreciable time. It's possible to implement a ring buffer without ISR lockouts (using check-and-retry if there have been updates) but I really don't think that's worth the hassle.

dpgeorge, 2017-03-30:
> The timeout doesn't fix my bug, and it goes away again when I put the ISR locks back in.

Ok, so there are definitely 2 bugs then.

> Those symptoms can happen if the receive pointer gets corrupted and ends up indexing into the transmit buffer. 

The tx buffer uses a ring buffer and I can't see where there would be issues with concurrent non-atomic access.

The rx buffer could have issues (it's not a ring buffer), but on inspection of the disassembly I can't see how it would have a problem.

@kentindell I believe you that there is a bug with the buffers but I need to be able to reproduce it, or at least know exactly what the issue is, so I don't apply empirical ad-hoc fixes.

Can you please try the following independent things:
- apply the ISR lockout just around the rx buffer code on line 494 (in USBD_CDC_Rx) and see if that fixes your issue (in that case it's just an issue with the rx, not tx)
- instead of doing ISR lockout, change the rx buffer code in USBD_CDC_Rx to read:
```
          // Copy byte from device to user buffer
          buf[i] = UserRxBuffer[UserRxBufCur];
          UserRxBufCur += 1;
```
and at the same time declare UserRxBufCur and UserRxBufLen as volatile (lines (73 and 74).

Thanks!

dhylands, 2017-03-30:
I took a look at the disassembly, and there is something suspicious in USBD_CDC_Rx.
This line:
```
buf[i] = UserRxBuffer[UserRxBufCur++];
```
is coded in the assembly as if this was written:
```
oldCur = UserRxBufCur;
UserRxBufCur++;
buf[i] = UserRxBuffer[oldCur]
```
If an ISR occurs between the increment and the buf[i] assignment and buf[i] is retrieving the last byte, then CDC_Itf_Receive might see that Cur == Len and move the buffer contents before buf[i] is retrieved.

Delcaring Cur & Len to be volatila and changing the code to put the increment after the buf[i] assignment:
```
        buf[i] = UserRxBuffer[UserRxBufCur];
        UserRxBufCur++;
```
shows the increment occuring after the retrieval of data.

kentindell, 2017-03-30:
Yes, that's one interleaving that could cause problems with Rx. The other one is in the increment itself:

1. Read ``UserRxBufCur``
2. ISR interrupts and ``CDC_Itf_Receive()`` runs, sees data has been read, sets ``UserRxBufCur`` and ``UserRxBufLen`` to 0, returns to interrupted code.
3. Code adds 1 to value read in (1).
4. Writes value back to ``UserRxBufCur``

Now we have the pointer not reset, but the length reset. This means that a subsequent read will run off the end of the buffer and start reading garbage further on in memory. Which will happen to be allocated by the linker to:

    static uint16_t UserRxBufCur = 0; // points to next available character in UserRxBuffer
    static uint16_t UserRxBufLen = 0; // counts number of valid characters in UserRxBuffer
    static uint8_t UserTxBuffer[APP_TX_DATA_SIZE]; // data for USB IN endpoind is stored in this buffer

So four bytes and then __the contents of the transmit buffer__. My [MIN protocol](https://github.com/min-protocol/min) code automatically skips over garbage, treating it as line noise, and resyncs to valid messages that pass CRC checks, which of course it finds left over in the transmit buffer. So my soak test code intermittently receives back what it sent: exactly explaining the symptoms I see.

On the transmit side I didn't look too hard before and just put the guards in. But looking closer I think it's free from interleaving problems.

dpgeorge, 2017-03-30:
> I took a look at the disassembly, and there is something suspicious in USBD_CDC_Rx.

Thanks @dhylands for looking into this.  I also looked at the disassembly and for the compiler I'm using (arm-none-eabi-gcc 6.3.0) I see a different pattern:
```
newCur = UserRxBufCur + 1;
data = UserRxBuffer[UserRxBufCur];
UserRxBufCur = newCur;
buf[i] = data;
```
and in that case there won't be any issues if the IRQ interrupts at any point in that code (because UserRxBufCur is not updated until after the data is read).

> 1. Read UserRxBufCur
> 2. ISR interrupts and CDC_Itf_Receive() runs, sees data has been read, sets UserRxBufCur and UserRxBufLen to 0, returns to interrupted code.
> 3. Code adds 1 to value read in (1).
> 4. Writes value back to UserRxBufCur

@kentindell I don't see how this is an issue: in point 2 the CDC_Itf_Receive function does not see that the data has been read because UserRxBufCur is unchanged (it's still less than UserRxBufLen).

kentindell, 2017-03-30:
If somehow ``UserRxBufCur`` gets to be bigger than ``UserRxBufLen`` then nothing stops the read loop from reading off the end of the buffer and into the transmit buffer. I'm pretty sure that this overrun is happening based on the way my soak test fails.

I've now managed to run the tests with the two variables declared as ``volatile`` and that seems to no longer trigger the problem (I've also added a test for an overrun with a ``printf()`` so if it does occur then at least we know for sure what has happened).

dhylands, 2017-03-30:
@dpgeorge Yeah - I was building with gcc 4.9.3. This is the assembler generated:
```
 8042998:	1c59      	adds	r1, r3, #1
 804299a:	8011      	strh	r1, [r2, #0]
 804299c:	4a05      	ldr	r2, [pc, #20]	; (80429b4 <USBD_CDC_Rx+0x64>)
 804299e:	5cd3      	ldrb	r3, [r2, r3]
 80429a0:	f805 3b01 	strb.w	r3, [r5], #1
```
The strh is saving the updated UserRxBufCur and the strb is saving into buf.

Probably not related, but USBD_CDC_RxNum could return incorrect results if an interrupt occurs between the fetch of Cur & Len.

Another possible scenario is that if Cur & Len are equal and greater than zero and the interrupt occurs between the loads in the statement:
```
while (UserRxBufLen == UserRxBufCur) {
```
and the interrupt sets Len and Cur back to zero, then the while loop would break when its not supposed to. Cur would then get incremented to 1, and since Cur != Len, it would continue to run copying data until the read buffer was full. This sounds like the scenario that @kentindell is running into.

kentindell, 2017-03-30:
@dhylands Yep, looks exactly right. And changing to ``volatile`` doesn't close off the race condition in the test: it probably just makes the window smaller, making the problem more intermittent (hence me not finding it).

kentindell, 2017-03-31:
Just an update. I declared the two Rx variables ``volatile`` and changed the loop test from:

    UserRxBufLen == UserRxBufCur

to:

    UserRxBufLen <= UserRxBufCur

This catches the race that @dhylands highlighted. I haven't had any problems since.

dpgeorge, 2017-04-04:
In 9a8e7f7a8e93e9319d2ff5795f4b8a6b9e4ba5ad I fixed the issue with the in-endpoint timing out too early.

In 3b447ede78970df04e08eeb1cc549a574a733667 I changed the RX code to use a circular buffer, which is a long-overdue change.  That should fix the races described above.  @kentindell are you able to check this new code to see if it fixes your issue?

dpgeorge, 2017-04-26:
I assume the above patches fixed this issue.

closed 2017-04-26 by dpgeorge