micropython/micropython #2985: "ussl.wrap_socket()" sockets has a different API than regular sockets

jcea, 2017-03-29:
```
MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266
Type "help()" for more information.
>>> import usocket, ussl
>>> s = usocket.socket()
>>> s.recv, s.send
(<bound_method>, <bound_method>)
>>> s.connect(('193.146.32.201', 443))
>>> s = ussl.wrap_socket(s)
>>> s.recv, s.send
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ussl' object has no attribute 'recv'
```

I know about "write()" and "read()", but I would expect the API to be the same.

pfalcon, 2017-03-29:
MicroPython doesn't work in terms of "sockets", it works in terms of "streams" (which is a protocol in Python-speak, usually called "interface" in other languages). Both objects above implement stream protocol. Normal socket also optionally implements legacy, idiosyncratic methods like send/recv, whcih aren't recommended for use in MicroPython (but may be used for compatibility with CPython). Summing up, the situation above is correct, and by design.

pfalcon, 2017-03-29:
Note that you can implement "ssl" module which wraps "ussl" module and provides any API you like.

closed 2017-03-29 by pfalcon