micropython/micropython #2986: SPI: mp_machine_soft_spi_init() should support NULL MISO or MOSI, and LSB

MrSurly, 2017-03-30:
It appears the intent in the file `extmod/machine_spi.c`, function `mp_machine_soft_spi_init` is that the `miso` and `mosi` options can be `None`, but in practice it fails because `mp_hal_pin_*` won't accept "not a Pin".  There are definitely use cases (e.g. my Sharp Memory LCD) that might be output-only or input-only.

Also, LSB transactions are useful, again for something like the Sharp Memory LCD.  Oddly, this display expects commands in MSB, and screen data in LSB.  Would it be weird if the LSB option were per transaction?

dpgeorge, 2017-03-30:
> It appears the intent in the file extmod/machine_spi.c, function mp_machine_soft_spi_init is that the miso and mosi options can be None

Not quite.  The intent is that if you don't specify the sck/miso/mosi keyword args then those pins are unchanged, but if you do specify them then they are, eg:
```python
spi.init(mosi=Pin(5)) # change mosi pin only
spi.init(sck=Pin(4), miso=Pin(6)) # change sck and miso pins only (leave mosi as-is)
```
Note that NULL is different to None.  The former is an internal C thing to be able to tell if you passed anything at all.

> There are definitely use cases (e.g. my Sharp Memory LCD) that might be output-only or input-only.

I agree, but this feature is not yet supported by any of the SPI drivers/bindings.  Probably the simplest thing to do is allow to pass None as a pin (mosi and/or miso) and if the driver supports it then it won't input/output on that pin.

MrSurly, 2017-03-31:
> Not quite. The intent is that if you don't specify the sck/miso/mosi keyword args then those pins are unchanged, but if you do specify them then they are

Ah! Okay, now it makes much more sense; I hadn't considered the use case of changing an existing SPI object. 

> Note that NULL is different to None. The former is an internal C thing to be able to tell if you passed anything at all

C I'm familiar with; I wasn't sure if `mp_const_none == NULL`, and you've just confirmed it is not.

> I agree, but this feature is not yet supported by any of the SPI drivers/bindings. 

Understood; part of writing this issue was to better understand how to implement the HW SPI on the ESP32 for μPy.

> Probably the simplest thing to do is allow to pass None as a pin (mosi and/or miso) and if the driver supports it then it won't input/output on that pin.

That's exactly what I've already implemented for ESP32, and it's already supported in `extmod/machine_spi.c`.  However, I need to implement the "re-configure existing SPI object" idiom, above.

closed 2017-03-31 by MrSurly