micropython/micropython #2991: undefined reference to `mp_module_usocket'

makerdiary, 2017-04-02:
Hi guys,
When I compile micropython for zephyr, it can not find `mp_module_usocket`.  Any ideas about this? 

```
/micropython/zephyr/libmicropython.a(objmodule.o):(.rodata.mp_builtin_module_table+0x3c): undefined reference to `mp_module_usocket'
/micropython/zephyr/libmicropython.a(objmodule.o):(.rodata.mp_builtin_module_weak_links_table+0xc): undefined reference to `mp_module_usocket'
collect2: error: ld returned 1 exit status
```

Thanks.

pfalcon, 2017-04-02:
I can't reproduce this. So, feel free to specify which git revision you try to build, how exactly you do that, etc. Also, you may want to check CI status of Zephyr port at https://ci.linaro.org/job/zephyr-aeolus/ (which is green as of now).

pfalcon, 2017-04-07:
Not reproducible, closing. Feel free to reopen if you have additional info.

closed 2017-04-07 by pfalcon