micropython/micropython #2996: wlan.scan('myssid')

pacmac, 2017-04-05:
Hi;

Presently, the only way I know to get the connection info (i.e. RSSI) is to perform a network scan() and then loop  through all the results searching for a known SSID and parse the values.

This takes valuable time when running on battery power and is not very efficient, it also requires that you need to pass the SSID to your function in order to extract the correct record, rather than being able to simply get the 'connected ssid'.

I notice that the API (wifi_station_scan) supports providing the ssid variable with the scan, i'e scan('my-ssid') however the micropython implementation does not. This may help speed up the scan process and reduce the scan time and complexity of the code needed to extract the required data.

Would it be possible to enhance the present design to:

1) Add the option of passing the ssid to the scan function: wlan.scan('my-ssid')
2) Add the option of scanning the connected SSID wlan.scan(wlan.CONNECTED_SSID)

Many Thanks

pfalcon, 2017-04-05:
RSSI question comes up not the first time, please search older tickets to see what previous discussion same to.

Anyway, with people asking for it, I don't remember anybody answering a simple question: is there ESP8266 API function which allows to:

> being able to simply get the 'connected ssid'.

If there's such call, let's work from there.

> Would it be possible to enhance the present design to:    Add the option of passing the ssid to the scan function: wlan.scan('my-ssid')

MicroPython implements a generic API, working the same way across all hardware. So, to answer that question, you would need to review whether it's possible to support that call on many different WiFi module models. You can start with WiFi modules already supported by uPy: e.g. Wiznet5000, CC3100/CC3100, ESP8266, ESP32, ATWINC1500. But that would be just a starting sample, to come to a definitive conclusion, you'd need to review many more models.

pacmac, 2017-04-05:
Thanks for the comprehensive reply.

I have checked before, and my conclusion was that there is no way to get the connection status directly from the connection and that a scan() is necessary.

I was just trying to make that scan() more efficient / faster by only scanning for the connected  SSID.

But I have to admit that I did not consider compatibility for all the other devices, would it not be possible to allow this option only if the device supports it ?

An alternative for devices that don't support this might be to scan for all ssids and filter / return the connected one by looping through the ssids, this would effectively be no slower that having to do the same thing in code on devices that don't support the feature but hopefully faster for those that do ?

As you know battery charge is precious, and just saving 2 seconds during a wake up can have a dramatic positive effect on the battery life ;-)

pfalcon, 2017-04-07:
@pacmac , let me start from the end:

> As you know battery charge is precious

If battery charge is precious, and scan is power-hungry, then perhaps not performing a scan can be a solution?

> I was just trying to make that scan() more efficient / faster by only scanning for the connected SSID.

What makes you think that scan for a particular SSID will be more efficient than a normal scan?

> But I have to admit that I did not consider compatibility for all the other devices, would it not be possible to allow this option only if the device supports it ?

I don't know, feel free to propose a scheme which would allow that, will be compatible with alternative proposals (for example, people wanted to be able to scan for hidden ssids), and ideally, compatible with any future requests/features.

> An alternative for devices that don't support this might be to scan for all ssids and filter / return the connected one by looping through the ssids, this would effectively be no slower that having [...]

Maybe. We need facts. Please feel free to experiment - first with esp8266, then with other devices, and share your results - preferrably in a way reproducible by anyone (e.g. if you claim that scanning for a particular SSID on esp8266 is twice as fast as a "nameless" scan, feel free to provide a script which can be easily run and will prove that by outputting times).

closed 2017-04-08 by pacmac