micropython/micropython #2997: Missing references in 1.8.7 release

dj095, 2017-04-05:
FYI i already got and built the esp-open-sdk toolchain (xtensa) and exported it for building micropython esp8266 module.
So i was building micropython for esp8266 from release 1.8.7 commit #5653e3c72fc8555c6a060acf6447ac694a036053
which is the latest release till date. 
I did cd micropython; make -C mpy-cross; git submodule update --init; cd esp8266; make axtls; make
All the commands work properly except the last 'make' command. I stumble onto this error :
Missing file "mpconfigboards.h". So i searched it on net and put it there. But there is another issue now. It says :
In file included from ../py/mpconfig.h:45:0,
                 from ../py/nlr.h:36,
                 from ../py/argcheck.c:30:
../stmhal/mpconfigport.h:288:10: error: #include expects "FILENAME" or <FILENAME>
 #include STM32_HAL_H

I know this #include is supposed to be replaced with <stm32f4xx_hal.h> or something like that. But it doesn't and i can't get this build to work. And i know i can get one of the existing builds out there. But i want to build the source code on my own. Please help.

dj095, 2017-04-05:
Okay. I removed the 'mpconfigboard.h' file i copied from net. And read the mpconfigport.h file properly and i understand now that its a generic file and so in the Makefile in esp8266 folder, i added : CFLAGS_EXTRA='-DMP_CONFIGFILE="<mpconfigport.h>" to tell it to use this config while building the ESP8266 port and did 'make -B'. And it got past that error. Now i'm stuck at mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
../py/mkrules.mk:73: recipe for target 'build/genhdr/qstr.i.last' failed
make: *** [build/genhdr/qstr.i.last] Error 2
Am i doing something wrong here?

dpgeorge, 2017-04-05:
It sounds like you have some basic issues with your build environment / command line.  Building esp8266 should work out-of-the-box.

> Missing file "mpconfigboards.h"

Is that the exact error message?  It doesn't look right, there's no such file needed by esp8266.

Please start again from scratch and specify your OS and give the exact error message you get.

dj095, 2017-04-05:
No the exact error message is : 
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
../py/mkrules.mk:73: recipe for target 'build/genhdr/qstr.i.last' failed
make: *** [build/genhdr/qstr.i.last] Error 2

dj095, 2017-04-05:
My OS: 
Mac OSX Sierra

dj095, 2017-04-05:
If the problem is OS Specific, i could try to build this whole thing again on a Linux Virtual Machine on vagrant or something like that.

pfalcon, 2017-04-05:
Yes, the supported build environment is Linux. if you have MacOSX problems, I'd recommend to open a topic at http://forum.micropython.org/viewforum.php?f=16 , you likely will get a reply sooner.

dj095, 2017-04-05:
Thanks. I will try to build it on virtualbox Linux. And there is one other thing i would like to ask you @pfalcon : 
1. I took a working copy of micropython ota build 1.8.7 from the downloads page of micropython. 
2. And flashed it onto my esp12F Module. It works fine by the way. 
3. Then i used the .ota file that was digitally signed with with same key as the micropython .bin file. 
4. Then i used your ota_client.py script and ran it with the .ota file when the ota-server was 
    listening(OTA Mode). The .ota file was signed with same digital signature as included in the config  
    file for the ota_server. So the server started picking up the digitally signed packages and after a  
    while  a reset happened due to the WatchDogTimer(WDT Reset). 
5. Then the rest of the packages didn't go through. It says "Unexpected offset". Of course the offset 
    that the ota_server expects now is 0 because it restarted. So I studied your code for ota_server 
    and found a line wdt_timer_feed() to keep the WDT fed.
6. I want to incorporate this line in your Server loop when its receiving UDP Diagram packets. So i 
    how can i incorporate this change such that its supported in the micropython esp8266 1.8.7 code 
     ? And will this change actually stop the WDT_Timer while receving packets?

dpgeorge, 2017-04-05:
Ok, it could be related to #2959.  The patch 23a693ec2d8c2a194f61482dc0e1adb070fb6ad4 may need to be reverted.

@dj095 does it work with a fresh repository?  Does it work if, on line 74 of py/mkrules.mk, you change the `$?` to `$^`?

dj095, 2017-04-05:
I tried it with $^ and now it gives :
GEN build/genhdr/qstr.split
Traceback (most recent call last):
  File "../py/makeqstrdefs.py", line 105, in <module>
    with open(args.input_filename) as infile:
IOError: [Errno 2] No such file or directory: 'build/genhdr/qstr.i.last'
../py/mkrules.mk:82: recipe for target 'build/genhdr/qstr.split' failed
make: *** [build/genhdr/qstr.split] Error 1

I guess i should try it on linux virtual machine now.

pfalcon, 2017-08-23:
Closing due to inactivity. If problem persists, feel free to open a new ticket with up to date info. Thanks.

closed 2017-08-23 by pfalcon