micropython/micropython #2999: Deepsleep changes GPIO State

pacmac, 2017-04-05:
This is also reported in the micropython forum.

I have completely isolated the ESP12S from all other components to eliminate the possibility of external effects and only connected TX, RX, VIN and GND.

Then I connected a DMM between GPIO13 and ground and executed the following code:

    import machine as mc
    p13 = mc.Pin(13, mode=mc.Pin.OUT)
    p13.high()

The DMM reads 3.3 Volts as expected.

    mc.deepsleep()

The DMM now reads 0 volts (or a few millivolts) so I can confirm that putting the device into deepsleep does appear to change the state of the GPIO, I also tried this on GPIO12 and the result is the same.

dpgeorge, 2017-04-06:
> This document appears to show that putting the ESP8266 device into deepsleep should have no effect (other than reducing the current to 2ua) on the GPIO states

Please can you point out the exact part of the document that makes such a claim.

The usual case with deep sleep modes is that everything shuts down, including GPIO peripherals.  In the document you link to it says "only the RTC module is still working" (sec 4.1).  So I assume the GPIO are all tristated during deep sleep.

diginfo, 2017-04-06:
Hi;

See the attached table
[image]

pacmac, 2017-04-06:
See attached Table
[image]

robert-hh, 2017-04-06:
Hello Pacmac. I've looked into the document you referenced, and that could indeed be understood in that the GPIO keeps it's level and is just reducing the sourcing strength. But your and my tests show that this is not the case. From my test it seems that the port is in a high impedance state then. So you should be able to pull it high with an external resistor.

pacmac, 2017-04-06:
@robert-hh - thanks so much for verifying my findings, I thought I was going crazy !

pfalcon, 2017-04-06:
That table can't be true (in a general definition of "true"), because there's a separate register for "output enable during sleep". Uncareful reader would issue an exclamation here, but careful would notice that the register is called "sleep", while you're talking about "deepsleep". Are they related or not? Your bet is as good as anybody else's.

Few conclusions (from a big number of cases like above):

1. Take Espressif documentation with a grain of salt.
2. In general, there're only 2 parties which can answer your questions: a) Espressif; b) you.
3. You're welcome to rely on community knowledge, but as an example, a moderately complex MCU from a reputable vendor has few thousands of pages in datasheet (example: MK64FN1M0VMD12 from NXP, 1789pp., and that merely has an Ethernet builtin). That's probably order of magnitude less than available for ESP8266, so 9 in 10 of your questions would go unanswered.

pacmac, 2017-04-06:
Also, reading that document section 4.1 - Features, it is re-stated that the GPIO state (should be) maintained:

    � Note:
    In Deep-sleep mode, GPIO maintains the state of its level and has a 2 μA drive strength.

chrismas9, 2017-04-06:
from the ESP8266EX datasheet rev 5.3

> • Deep-sleep: Only RTC is powered on – the rest of the chip is powered off. Recovery
memory of RTC can save basic Wi-Fi connection information.

> For low power operations, the GPIOs can also be set to hold their state. For instance, when
the chip is powered down, all output enable signals can be set to hold low.
Optional hold functionality can be built into the IO if requested. When the IO is not driven by
the internal or external circuitry, the hold functionality can be used to hold the state to the
last used state. The hold functionality introduces some positive feedback into the pad.
Hence, the external driver that drives the pad must be stronger than the positive feedback.
The required drive strength is small — in the range of 5 μA to pull apart the latch.

If the rest of the chip (except RTC) is powered off the Output Registers will be unpowered and unable to drive the GPIO or enable the pin hold function. If GPIO's hold their state during deep sleep, it will be because of the hold function (low current positive feedback). However it seems this is not enabled by default. I can't find a register definition for hold enable, and anyway the control registers will be powered down. So this looks a bit like the high accuracy 32 kHz RTC oscillator - vapourware. The datasheet doesn't say how to request the optional hold function.

The only way the hold circuit could work is if it was always active, even in input (tristate) mode. If the hold logic is active a 1M load (eg multimeter) might flip the hold circuit to 0 and that would explain reading close to 0 V rather than 2V. 

A 10 Mohm input multimeter or Oscilloscope should give a reliable reading but lower impedance devices may not.

dpgeorge, 2017-05-28:
This is more of a hardware issue than software.  If it can be shown (by experiment/testing) that the esp8266 can indeed hold a GPIO state during deepsleep, and that MicroPython doesn't control that feature correctly in its software, then please reopen.

closed 2017-05-28 by dpgeorge

rumpeltux, 2019-09-05:
For everyone else coming here searching about this issue:
the [updated datasheet (v1.5)](https://www.espressif.com/sites/default/files/documentation/9b-esp8266-low_power_solutions__en.pdf) (table 1-1) now officially mentions that GPIO state in deep sleep is undefined. (The 2μA number still seems wrong from my measurements).