micropython/micropython #3000: ESP8266 Sleep Not Implemented

chrisyokum, 2017-04-05:
On the ESP8266:
>>> machine.sleep()                                                                                                                         
Warning: not yet implemented                                                                                                                
>>>                                                                                                                                         
I don't need the full deepsleep savings, what's saved by just using the milder sleep modes will be sufficient.

chrisyokum, 2017-04-05:
I found that if I issue esp.sleep_type(esp.SLEEP_LIGHT) - this is implemented, and seems to slow the REPL responsiveness a little, but I don't notice any difference in actual power consumption. With either SLEEP_MODEM (the default) or SLEEP_LIGHT enabled, if I then execute while True: machine.idle() the power consumption drops by about 10ma but it's still >100ma overall.

dpgeorge, 2017-04-07:
The esp8266 does not have very good low-power operating modes, so this is really a hardware issue and not something we can solve in software.

The machine.sleep() function is supposed to put the machine into state where it stops processing but retains all state (eg RAM, registers) and when it's woken it continues executing the code just after the sleep() call.  How much the power consumption is reduced by during sleep() is up to the device/MCU (also how it is woken is device specific).

If you know a way of making the esp8266 perform such a "light sleep" that conforms to the above behaviour then please let us know and we can look to implement that.

EternityForest, 2017-04-07:
Doesn't micropython enable AP mode by default, even when you also turn on STA mode? IIRC AP mode prevents sleep from working as expected because it has to be always listening for messages from clients.

peterhinch, 2017-04-08:
You can turn off AP mode.

pfalcon, 2017-04-16:
@chrisyokum : The standard recommendation for people who really want to save power is to use deep sleep. People who don't want to save as much power as possible will find the results that you have found. If you're interested to get to the bottom of it, I encourage you to join ESP8266 reverse engineering project - with enough people working daily on it, in few years, we will be able to make it work like we want, not like it does (maybe, because WiFi power saving largely depends on the network, not on a particular module in it).

chrisyokum, 2017-04-17:
Thanks for the advice everybody! It turns out I did have AP mode enabled, but even disabling that (and confirming via a network monitor) didn't bring it down. I think I agree with pfalcon that you use deep sleep if you really care.. I'm going to put this off until I have some more time to focus on this.

girtsf, 2017-07-25:
I was trying to get some of the modem or light sleep modes working and I did not succeed either.

Doing
```
wifi_set_opmode_current(NULL_MODE);
wifi_fpm_set_sleep_type(MODEM_SLEEP_T);
wifi_fpm_open();
wifi_fpm_do_sleep(0xFFFFFFF);
```
is not enough. The SDK docs say 
> When forced sleep interface is called, the chip will not enter sleep mode instantly, it will enter sleep mode when the system is executing idle task.

This is where I'm a bit lost. Is idle task something that is implemented in the SDK, but not being used by micropython because of the custom main loop implementation in `ets_alt_task.c`?

The hardware is definitely capable of doing this: if I flash a recent build of nodemcu lua firmware and issue `wifi.suspend({duration=0})`, current goes from ~50mA to ~0mA, and the device still responds to commands on its REPL.

The relevant code for nodemcu is here:
https://github.com/nodemcu/nodemcu-firmware/blob/3e24b1c75b4b4fd469be784fd5d596f0eb3871ff/app/modules/wifi.c#L357
https://github.com/nodemcu/nodemcu-firmware/blob/3e24b1c75b4b4fd469be784fd5d596f0eb3871ff/app/pm/pmSleep.c#L336

jarekdudzinski, 2017-08-06:
Any update on implementing that sleep modes?

one-quaker, 2018-10-28:
How about deep sleep at this moment?

dpgeorge, 2019-05-21:
`machine.lightsleep()` was implemented on the esp8266 in 808dc95ab80d2e6be9a0a602d65cdee35903fc25, and other power saving measures were added so that it has much lower power consumption in certain cases.

closed 2019-05-21 by dpgeorge