micropython/micropython #3001: Having issues sending a float to and getting a float from a custom module.

cody-hanks, 2017-04-06:
This is just a random function to make sure that I understand the process of exposing a function to the python code. 

my function is 
```
STATIC float addten(float val){
    return (float)(val +10.0);
}
STATIC MP_DEFINE_CONST_FUN_OBJ_0(addten_obj,addten);
STATIC const mp_map_elem_t globals_table[]={
     {MP_OBJ_NEW_QSTR(MP_QSTR_addten),(mp_obj_t)&addten_obj },
};
```

The error message I am seeing is, 
     ***.c:21:1: error: initialization from incompatible pointer type

I have tried with the float32_t, float, and mp_float_t for return type .... 

any chance some one can send me in the right direction

dpgeorge, 2017-04-06:
The correct thing to do is
```c
STATIC float addten(mp_obj_t val_in){
    float val = mp_obj_float_get(val_in);
    return mp_obj_new_float(val + 10.0);
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(addten_obj,addten);
STATIC const mp_map_elem_t globals_table[]={
     {MP_OBJ_NEW_QSTR(MP_QSTR_addten),(mp_obj_t)&addten_obj },
};
```

> If this is the wrong place to ask please let me know and I can move this somewhere else.

For this kind of question, in the future please use https://forum.micropython.org

closed 2017-04-06 by dpgeorge

peterhinch, 2017-04-06:
@cody-hanks For FFT you might be interested in [this](https://github.com/peterhinch/micropython-fourier).