micropython/micropython #3004: Assertion failure when passing message containing '%' to mp_obj_new_exception_msg

stinos, 2017-04-06:
Passing a message which contains the `%` character causes the `assert(*fmt == '%' || !"unsupported fmt char");` assert in mpprint.c because mp_obj_new_exception_msg just forwards to mp_obj_new_exception_msg_varg and that one passes anything containing `%` through a printf - which is not needed/wanted in this case. This was probably done to reuse the implementation, but also means messages containing `%` cannot be used with any of the widely used mp_obj_new_exception_msg/mp_raise_msg functions.

This is low priority but it's a bit inconvenient when re-raising messages from external libraries which can contain anything in their error message (figured this out using the pyserial library)..

pfalcon, 2017-04-06:
Is this triggerable from Python side?

stinos, 2017-04-06:
Not that I know of, I think all occurrences where it gets used in uPy itself contain hardcoded error strings (with no % character).

dpgeorge, 2018-12-06:
Thanks for reporting this, since upon investigation it showed that static error messages passed to `mp_obj_new_exception_msg()` are unconditionally copied to a new buffer on the heap, which is really a waste of heap memory since the message string data could be referenced directly without making a duplicate.  Arguably exceptions with messages don't usually happen during the course of normal execution, so the unnecessary heap allocation is not that bad, but still it would be nice to fix it.  The fix would then also fix the issue here, which really is a bad bug because printf should not be called on strings that are not supposed to be formatted.  See #4335 for a fix.

dpgeorge, 2018-12-10:
Fixed by 55830dd9bf4fee87c0a6d3f38c51614fea0eb483

closed 2018-12-10 by dpgeorge