micropython/micropython #3005: uasyncio stream reader/writer: unexpected behaviour with UART

peterhinch, 2017-04-06:
Testing done on Pyboard V1.1. The ``sender`` coroutine iterates every 2 seconds but no output appears on pin X1:
```python
import uasyncio as asyncio
from pyb import UART
uart = UART(4, 9600)

async def sender():
    swriter = asyncio.StreamWriter(uart, {})
    while True:
        swriter.awrite('Hello uart\n')
        await asyncio.sleep(2)
        print('wrote')

loop = asyncio.get_event_loop()
loop.create_task(sender())
loop.run_forever()
```
The next script produces output on X1 as expected until a loopback is applied (link X1 and X2). At this point the UART sends data continuously. The data is read correctly, but the call to

``await asyncio.sleep(2)`` no longer behaves as expected.
```python
import uasyncio as asyncio
from pyb import UART
uart = UART(4, 9600)

async def sender():
    while True:
        uart.write('Hello uart\n')
        await asyncio.sleep(2)
        print('Wrote')

async def receiver():
    sreader = asyncio.StreamReader(uart)
    while True:
        res = await sreader.readline()
        print('Recieved', res)

loop = asyncio.get_event_loop()
loop.create_task(sender())
loop.create_task(receiver())
loop.run_forever()
```

pfalcon, 2017-04-06:
As expected, awrite() is a coroutine and must be awaited.

peterhinch, 2017-04-06:
Ooops. Apologies for that 'senior moment' on my part.

Alas the second script still behaves oddly, even adapted as below with ``sender()`` corrected to use ``awrite()``. With the loopback missing pin X1 correctly issues serial data at two second intervals. As soon as X1 and X2 are linked the data stream as seen on a 'scope becomes continuous, with the program outputting
```
wrote
Recieved b'Hello uart\n'
```
continuously and at high speed, rather than at two second intervals.
```python
import uasyncio as asyncio
from pyb import UART
uart = UART(4, 9600)

async def sender():
    swriter = asyncio.StreamWriter(uart, {})
    while True:
        await swriter.awrite('Hello uart\n')
        await asyncio.sleep(2)
        print('wrote')

async def receiver():
    sreader = asyncio.StreamReader(uart)
    while True:
        res = await sreader.readline()
        print('Recieved', res)

loop = asyncio.get_event_loop()
loop.create_task(sender())
loop.create_task(receiver())
loop.run_forever()
```

pfalcon, 2017-04-06:
Did you try to see if anything in https://github.com/micropython/micropython/issues/2622 (like, a known issue) is relevant.

Otherwise, to set expectations right: https://github.com/micropython/micropython-lib/pull/164/ is the high-priority bug. After it, there's high-priority task of implementing poll.ipoll() for baremetal and switch to it. After that the original work on optimizing (networking) uasyncio can continue again. Which means next few month you'd be ~ the only one to look at uart use, different scheduling priorities, etc. That's of course an encouragement to do that ;-).

dpgeorge, 2017-04-07:
I tested the improved sender/receiver code above and indeed the bug (premature scheduling of the sender due to receiver having data) is exactly that which is fixed by https://github.com/micropython/micropython-lib/pull/164

dpgeorge, 2017-04-07:
Just to say it explicitly: the issue reported here is fixed by https://github.com/micropython/micropython-lib/pull/164

peterhinch, 2017-04-07:
> Just to say it explicitly: the issue reported here is fixed by micropython/micropython-lib#164

Yes, now fixed.

closed 2017-04-07 by peterhinch