micropython/micropython #3006: wlan status() states not defined

chrisyokum, 2017-04-06:
The documentation for the network module talks about STAT_IDLE, STAT_CONNECTING, etc as defined constants that may be returned by the status() function, however these constants don't seem to be defined:

>>> sta_if.status()                                                                                                                                   
5                                                                                                                                                     
>>> sta_if.STAT_IDLE                                                                                                                                  
Traceback (most recent call last):                                                                                                                    
  File "<stdin>", line 1, in <module>                                                                                                                 
AttributeError: 'WLAN' object has no attribute 'STAT_IDLE'                                                                                            
>>> sta_if.STATE_IDLE                                                                                                                                 
Traceback (most recent call last):                                                                                                                    
  File "<stdin>", line 1, in <module>                                                                                                                 
AttributeError: 'WLAN' object has no attribute 'STATE_IDLE'                                                                                           
>>>

dpgeorge, 2017-04-06:
Try the network module itself: `network.STAT_IDLE`

chrisyokum, 2017-04-17:
That worked! Thanks! It's not clear in the documentation.

closed 2017-05-14 by pfalcon

hansonap, 2019-03-06:
Where did they go now?

network.STAT_IF results in "module object as now attribute STAT_IF".

New 1.10 documentation returns no results for "STAT_IF". The WLAN page talks about "STAT_*" but does not provide information about where they are stored.

dpgeorge, 2019-05-09:
@hansonap it depends which port you use as to what constants are available.  But note that `STA_IF` is distinct from `STAT_xxx`.