micropython/micropython #3007: ESP8266 virtual timer accuracy?

mika-koivusaari, 2017-04-07:
What should be the accuracy of  virtual timers?

Following code:
```import machine
import time

def callback(temp):
    print(machine.RTC().datetime())

tim=machine.Timer(-1)
tim.init(period=60000, mode=machine.Timer.PERIODIC, callback=callback)
```

Gives output:
```
MicroPython v1.8.4-65-g67d52d8 on 2017-02-10; ESP module with ESP8266

(2000, 1, 1, 5, 0, 0, 54, 33)
(2000, 1, 1, 5, 0, 1, 47, 722)
(2000, 1, 1, 5, 0, 2, 41, 383)
(2000, 1, 1, 5, 0, 3, 35, 116)
(2000, 1, 1, 5, 0, 4, 28, 920)
(2000, 1, 1, 5, 0, 5, 22, 737)
(2000, 1, 1, 5, 0, 6, 16, 487)
(2000, 1, 1, 5, 0, 7, 10, 241)
(2000, 1, 1, 5, 0, 8, 3, 950)
(2000, 1, 1, 5, 0, 8, 57, 745)
(2000, 1, 1, 5, 0, 9, 51, 552)
(2000, 1, 1, 5, 0, 10, 45, 358)
(2000, 1, 1, 5, 0, 11, 39, 195)
(2000, 1, 1, 5, 0, 12, 33, 131)
(2000, 1, 1, 5, 0, 13, 27, 119)
```

Which is about  6 seconds off.

pfalcon, 2017-04-07:
You of course wanted to ask first what's the accuracy of esp8266's "RTC" you use to measure timer accuracy, didn't you?

mika-koivusaari, 2017-04-07:
Tested with timing from a computer, and it indeed seems that timer is correct and rtc is way off.

I knew that esp's rtc wasn't that accurate but didn't realise it was that wrong, well, just have to deal with it.

closed 2017-04-07 by mika-koivusaari

robert-hh, 2017-04-08:
Since I was interested for another purpose, I compared the time counted by ticks_ms()/ticks_diff() against the RTC clock and time.time(). The reference was my PC's clock application. 
Within 2 hours I could not see any deviation to the reference clock for ticks_ms(), meaning that the deviation was <1s, but time.time() was off by 152 s within than 7200 seconds!