micropython/micropython #3012: I2C address sensor

APOON24, 2017-04-10:
Hi everyone,
I'm trying to type a driver for the XNucleo (http://www.st.com/en/ecosystems/x-nucleo-iks01a1.html )sensor on the Pyboard. There are 4 sensor, humidity and temperature sensor, compass, pressure and accelerometer. These sensor are working on the I2C protocol and SPI.
At the beginning of my code, I'm initializing the I2C and scanning the bus. When I'm scanning the bus I receive 4 addresses but those addresses are not matching with the datasheet. Anyway I have tried to check the ID of each sensor with the 4 addresses before and I can just match 2 sensors. So something is not working well or I don't understand something.
Here my code on the REPL
from pyb import I2C
```
>>> com=I2C(2,I2C.MASTER,baudrate=100000)
>>> com.scan()
[30, 93, 95, 107]
```
then I read the register(who I am address=0x0f)
```
>>> com.mem_read(2,30,0x0F)
b'=\x14'
>>> com.mem_read(2,93,0x0F)
b'=\xbd'
>>> com.mem_read(2,95,0x0F)
b'=\xbc'
>>> com.mem_read(2,107,0x0F)
b'=\x00'
```
But only the sensors with the address 93 and 95 are good.

Someone can help me to understand this issue ? 

Thank you in advance
Alex Poon

robert-hh, 2017-04-10:
Hello Alex, this is quite a set of sensors. And you will certainly find people besides me who respond on the micropython forum https://forum.micropython.org/, which is the place dedicated for such questions.
At least a first glance in the data sheets told me, that the addresses [30, 93, 95, 107] are the ones to be expected. Therefore you might have to dig further into the description of each sensor. On the foruim might be people, who dealt with them before.

dpgeorge, 2017-04-10:
Since the I2C scan is returning 4 devices, that indicates that the bus and I2C driver are working.  Remember that I2C addresses in MicroPython are 7-bit addresses and sometimes datasheets list an 8-bit version of this address (which is just the 7-bit address times by 2).  Also, that XNucleo board allows you to select the low bit of the (7-bit) I2C address via the SD0 pin, so you should check what the value of SD0 is for each sensor.

@APOON24 if you find a bug in MicroPython please report it here, otherwise please use the forum for general questions.  Thanks!

closed 2017-04-10 by dpgeorge