micropython/micropython #3014: RFC: Ideas on unit-like testing of networking functionality

pfalcon, 2017-04-11:
Off-shotting this discussion from https://github.com/micropython/micropython/pull/2591 :

> This can only be done by using threads or separate scripts.

In general case, yes, but depending on the amount of data and complexity (determinism) of the exchange, it doesn't have to be, for *POSIX-compliant* sockets e.g.:

~~~
import socket

s1 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

s1.bind(("192.168.1.35", 32323))

s2 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

s2.sendto(b"foobar", ("192.168.1.35", 32323))

print(s1.recv(1000))
~~~

Now you just may wonder how much of such "corner" features embedded IP stacks have, and indeed, I found that for Zephyr such way currently doesn't work: https://jira.zephyrproject.org/browse/ZEP-1966

pfalcon, 2017-04-11:
Ok, the issue above was promptly fixed in Zephyr (and fix was easy, 2-line literally), so I could proceed with my idea: https://github.com/pfalcon/micropython/commit/d6477a9b0b29c11783a6234ccbddd4274048f5b1

However, that fixed only UDP case, and I found that TCP server connection doesn't work locally in Zephyr - packets are routed, but TCP connection handling apparently don't anticipate having both server and client locally, and is thus misbehaves.

dpgeorge, 2017-05-06:
Regardless of how network tests are run, in the meantime we shouldn't lose track of tests that should be part of the network test suite, such as: https://github.com/micropython/micropython/issues/2931#issue-211615859

dpgeorge, 2017-05-28:
It would be good to work out a way to move forward with this, ie defining network tests.  In particular so that issues like #2930, #2947, #2948 have a test for them, can be fixed, and then not regress (and so other ports with networking can be sure they have the correct behaviours).

Accepting that not everything can be tested using a loopback, ther will need to be a server of some kind that can respond to the tests.  In some cases it's as simple as having a live HTTP(S) server, but also UDP needs testing.  Ideas for having a "network test server" are:
- use existing protocols (eg HTTP, HTTPS, NTP) and live servers (eg micropython.org) and write tests that use these
- have a custom "network test script" running somewhere on the internet, eg on micropython.org
- have a custom "network test script" that must be run locally (ie within the local network, so no internet connection is needed)

The latter 2 options could be combined, using the same script, which could be left running on micropython.org for easy access, but also with the possibility to run locally if needed.  Of course run-tests is going to need a way to inject the IP address of the network server into the network tests.

pfalcon, 2017-05-31:
First of all, I guess it makes sense to decide whether it's worth pursuing "loopback" tests. The big benefit is that they can be run as normal unit-tests without any special test framework setup. But beyond trivial tests (like, calling socket() and checking that it succeeds), this may require special setup (loopback interfaces enabled, etc.) on device side, which may be not always enabled in production setup.

So, for more complete testing, we definitely will need "client-server" setup, where part(s) of the test runs on the host, and part(s) - on device. But how I imagined it is not a static "network test script", but rather each individual test case would include code snippet(s) to be run on the host, and other snippets - to run on the 
device (e.g. via pyboard).

Having just just one static "test server" won't be too flexible or detailed, that can be integration testing, but we should get as close as possible to unit testing.

And yep, I'd say it goes without a saying that default setup should involve communicating with DUT (device under test) and host, no internets involved. Or it will be even more complicated setup with even higher error rate.

That's partly speculative, maybe if I'm shown a single "network test script" which can be tested against and offers a true unit-test level selectively, I'll agree it's ok. But I already developed "loopback" style tests for Zephyr, hit its bugs which didn't allow to do further loopback testing (e.g. for TCP), and was ready to try "split host/DUT" approach, what stopped me is that serial communication in Zephyr was either not reliable enough to run such tests (fix should be available now for real hardware, but qemu is still inconclusive).

pfalcon, 2017-06-06:
Ok, I guess let's give this a try with a simple dumb static server (e.g. echo) and see where that leads us (be ready to change it).

Tests can be encoded using "test" function which accepts server address to connect to. Then for host testing, they can be imported and the function called with needed param, for device testing, pyboard.py can append a line which calls the function with needed param.

@dpgeorge , if you have any workings on that, please share/suggest, otherwise, I'm ready to start on that, as I need it for Zephyr testing.

dpgeorge, 2017-06-07:
> First of all, I guess it makes sense to decide whether it's worth pursuing "loopback" tests.

A loopback test is pretty much just going to be a unit test for the loopback feature itself, since some targets (eg cc3200) are never going to have loopback capabilities.

> if you have any workings on that, please share/suggest,

No I don't have anything.

dpgeorge, 2017-06-21:
Some preliminary network tests were added in the tests/net_inet/ directory.

pfalcon, 2017-06-23:
> Some preliminary network tests were added in the tests/net_inet/ directory.

Following the discussion in https://github.com/micropython/micropython/commit/458cbacb8f62ad0f33024360af61fb797530c65b#commitcomment-22688106 , these tests were actually moved to `tests/net_hosted/`, in 3f9d59c87a6797857a074737034bcd58d0726b2e, to serve as seed material for ideas expressed above (https://github.com/micropython/micropython/issues/3014#issuecomment-306563372).

pfalcon, 2017-06-23:
Further, bc7659eb59f8e7533a253475d4ca0ef26e1b2fc7 shows the idea how such tests would be written.

dpgeorge, 2020-06-09:
Network tests with multiple devices involved can now use `run-multitests.py`, see eg 8f44c0dd16df33fb0e3e9de1701787fdd0d67e57

closed 2020-06-09 by dpgeorge