micropython/micropython #3022: Create another new Timer interrupt inside handler ?

hieuhotxac, 2017-04-14:
```
import machine
from machine import ADC,Timer
import array
import network
import socket
import json

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect('Kim','0938860714')
time.sleep(2)
s = socket.socket()
s.connect(('192.168.1.3', 4000))
time.sleep(2)
s.send(b"living_room")

adc = ADC(0)
b = [ 1, 0, 0.00999539]
a = [ 1, 0.00400895, 0.00999539]
x = [0, 0, 0]
y = [0, 0, 0]
pack = array.array('int')

class read():
	def __init__(self):
		self.pack = []
		self.tim0 = Timer(0)
		self.tim0.init(period=2, mode=Timer.PERIODIC, callback=lambda t:self.readc())

	def readc(self):
		self.x = adc.read()
		self.state0 = machine.disable_irq()    #disable Interrupt of Timer 0
		x[2] = x[1]
		x[1] = x[0]
		x[0] = self.x
		self.y = fx()
		y[2] = y[1]
		y[1] = y[0]
		y[0] = self.y
		if y > ...:
			self.tim1 = Timer(1)                      #set new Timer Interrupt inside handler
			self.tim1.init(period =1, mode = Timer.PERIODIC, callback=lambda t:self.readraw())
			self.tim2 = Timer(2)
			self.tim2.init(period =..., mode = Timer.ONE_SHOT,callback = lambda t: self.stop())
		else:
			machine.enable_irq(self.state0)

	def readraw(self):
		x = adc.read()
		self.state1 = machine.disable_irq()
		pack.append(x)
		machine.enable_irq(self.state1)

	def stop(self):
		self.tim1.deinit()
		self.tim2.deinit()
		s.send(json.dumps(pack))
		del pack
		pack = array.array("int")
		machine.enable_irq(self.state0)

READ = read()
```
Is it posible ? After disable the main Timer Interrupt, I create two new Timer interrupt inside the handler. When everything is done, I deinit Timers I have created, and reactivating main Interrupt.

peterhinch, 2017-04-15:
I suggest you read [this](http://docs.micropython.org/en/latest/pyboard/library/machine.html#machine-callbacks). Timer instantiation and appending to arrays involve RAM allocation.

Queries such as this should be raised in the forum: this is a bug tracker.

pfalcon, 2017-04-16:
Hopefully the above clarifies it.

closed 2017-04-16 by pfalcon

hieuhotxac, 2017-04-16:
Thanks you. I will simplify my code.