micropython/micropython #3023: Problem with round on esp8266

slzatz, 2017-04-15:
MicroPython v1.8.7-623-g209eaec-dirty on 2017-04-15; ESP module with ESP8266

I haven't figured out the  pattern (since some results are correct) but there appears to be a problem with round.

>>> round(0.55, 1)
0.5
>>> round(0.56, 1)
0.5999999

dhylands, 2017-04-16:
Those both look like they could be correct to me. You have to keep in mind that its single precision floating point which is six significant digits. 0.5999999 with six significant sigits is the same as 0.6.

0.55 doesn't have an exact representation in base 2 (and floating point is all base 2), so it depends on whether 0.55 gets parsed as 0.55 + a small number or 0.55 minus a small number.

Now having said all that, on the pyboard I get: [code]>>> round(0.55, 1)
0.6
>>> round(0.56,1)
0.6[/code] so there is definitely a discrepancy.
You should read this: https://docs.oracle.com/cd/E19957-01/806-3568/ncg_goldberg.html

pfalcon, 2017-04-16:
Duplicate of https://github.com/micropython/micropython/issues/2616. Please search issues before submitting!

closed 2017-04-16 by pfalcon

slzatz, 2017-04-16:
and in cpython, 

round(0.55, 1)
0.6

dhylands, 2017-04-16:
> and in cpython,
> 
> round(0.55, 1)
> 0.6

CPython uses double floats (64-bit) so it really isn't comparable.