micropython/micropython #3024: Deconditionalize the documentation

pfalcon, 2017-04-16:
As an offshot of https://github.com/micropython/micropython/issues/2578 , this ticket is to track deconditionalizing of MicroPython documentation, making it generic for any port/board.

pfalcon, 2017-04-16:
There has been great progress over last couple of weeks:

~~~
$ grep -F ".. only" [^ip]*.rst | wc
     24      74     842
~~~

Which at least halved conditionalisms as they were before (the command above skips index.rst and pyb.* which are "ok" to contain conditions).

The specific list as of now:
~~~
machine.I2C.rst:.. only:: port_wipy
machine.I2C.rst:.. only:: port_wipy
machine.I2C.rst:.. only:: port_wipy
machine.I2C.rst:.. only:: not port_wipy
machine.I2C.rst:.. only:: port_wipy
machine.I2C.rst:.. only:: port_esp8266
machine.rst:.. only:: port_wipy
machine.rst:.. only:: port_wipy
machine.rst:.. only:: not port_wipy
machine.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.UART.rst:.. only:: port_pyboard
machine.UART.rst:.. only:: port_wipy
machine.UART.rst:.. only:: port_wipy
network.rst:.. only:: port_wipy
network.rst:.. only:: port_pyboard
network.rst:.. only:: port_esp8266
network.rst:.. only:: port_wipy
uos.rst:.. only:: port_pyboard
uos.rst:.. only:: port_wipy
uos.rst:.. only:: port_wipy
~~~

dpgeorge, 2017-04-18:
pyboard/stmhal doesn't (yet) have machine.UART so I removed it's conditional reference from machine.UART.rst in fabaa6143745cf09928dfc13367ddc91e0eb9ad2

dpgeorge, 2017-04-18:
cc3200's I2C class has now been changed to conform to the new HW API, and all I2C methods have been deconditionalised in 1f1a03d0c39f3b733372e523bc4906a2b40cfff0

pfalcon, 2017-04-18:
Great, thanks!

pfalcon, 2017-06-03:
Current status:

~~~
$ grep -F ".. only" [^ip]*.rst | wc
     16      49     547
~~~

~~~
machine.rst:.. only:: port_wipy
machine.rst:.. only:: port_wipy
machine.rst:.. only:: not port_wipy
machine.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.Timer.rst:.. only:: port_wipy
machine.UART.rst:.. only:: port_wipy
machine.UART.rst:.. only:: port_wipy
network.rst:.. only:: port_pyboard
network.rst:.. only:: port_esp8266
network.rst:.. only:: port_wipy
uos.rst:.. only:: port_pyboard
uos.rst:.. only:: port_wipy
uos.rst:.. only:: port_wipy
~~~

pfalcon, 2017-06-03:
Going to deconditionalize uos module by moving "Port specifics" section to ports' reference.

pfalcon, 2017-06-03:
1ce44f30467e4882580ca8545f15ba7d882cb428

dpgeorge, 2018-07-18:
All conditionals in pyb module and its classes were removed in 805fd0cfe6880153b990acc19f547e740b87f23d

Some wipy conditionals were removed from the machine module and machine.UART class in 4cc65e22d4464917d995dc89023ca8883c7dc6b2, 163cc66a0b4b8f305d1f9e11625cee4b5d2f75e2 and 3e0d587a496d41eca8911938781837d53d1fa17b

dpgeorge, 2018-07-20:
Conditionals were removed from the library index in 84d5dd46fec2aa5a7c585e42927178acf2055405, the reference index in 0ab84289952a9a9b7f35a89943f1e8e1ddd11696, and the machine module index in 81e320aecc2c7e21235244ade084ba4f71ff40e7

pfalcon, 2018-12-06:
> conditionals were totally removed from the top level, and build system, for the docs.

This was done, but https://github.com/micropython/micropython/blob/master/docs/README.md still describes building port-specific docs:
~~~
make MICROPY_PORT=<port_name> html
~~~
etc.

dpgeorge, 2018-12-10:
> This was done, but https://github.com/micropython/micropython/blob/master/docs/README.md still describes building port-specific docs:

Ok, fixed by beeeec292b2da48299d51c15039c0cc802fd7394