micropython/micropython #3030: Generalizing "soft reboot" feature to cover full reset possibility

pfalcon, 2017-04-19:
pyboard.py uses "soft reboot" feature to provide a clean environment for code execution. run-tests uses it to provide clean environment for each test. It's great to have this feature, without relying on the underlying hardware capabilities for reset (which may be not there, which may have various issues and artifacts, not work as expected, etc.). However, over last time, we faced the limitations of "soft reboot" approach too - e.g., on esp8266 it doesn't reinitialize network stack and other subsystems, which can lead to various problems either.

And as we start to approach automated network testing, e.g. https://github.com/micropython/micropython/issues/3014, that limitation becomes clear and disabling. For example, Zephyr network stack has enough bugs that without resetting it completely between tests, any results not just can, but are skewed.

So, it would be nice to have an option to implement "soft reboot" as "hard reboot". Per the initial paragraph, that would have various issues in general, but if it works in a particular case, why not?

And here we come to the essence of the issue: currently, REPL code relies that if soft reboot was done in raw REPL, then it remains in raw REPL. Of course, this is not attainable with hard reboot, so I'm proposing to change that - after soft reboot, the interpreter comes back with the normal REPL.

pfalcon, 2017-04-19:
Some trivia:

The testsuite runtime with hard reboot Zephyr qemu_x86:
~~~
real	1m8.854s
~~~
soft reboot:
~~~
real	0m32.768s
~~~

dpgeorge, 2020-06-09:
Closing due to inactivity.  And Zephyr now has better networking support so this issue is less of a problem.

closed 2020-06-09 by dpgeorge