micropython/micropython #3033: Any plans to add ATOM Plugin compatibility like Pymakr for general Micropython?

mohpor, 2017-04-20:
Hi,

I've been a user of Pycom's WiPy boards for a while and long story short, I've had to fall back to ESP8266 and I want to continue developing on Micropython.
There are a lot of IDE's available for Python, I use Atom.

Today I have noticed Pycom has released a Plugin for Atom which supports REPL and uploading code, ...

I guess it might be the time to add compatibility for the Upstream Micropython project with a decent IDE. Atom is a great candidate but personally, I'm open to any form of integrated, easy to use and understand environment to help me do more coding.

pfalcon, 2017-04-20:
> I guess it might be the time to ...

If you think so, please start prototyping code, show it to other users on the forum, get together with them (because you'll soon find out that alone you won't have much progress), elaborate your code with them, get feedback from hundreds of users. At which point it may indeed be right time to contact the upstream project and see what to do about it - whether it picks it up (and you become an upstream developer), or you keep maintaining it on your own.

mohpor, 2017-04-20:
@pfalcon 
Thanks for the encouragements, I really appreciate it.
Sounds like a good plan, I'm trying to survive a very harsh deadline on my end, but I'll make sure to try and make some effort very soon.

Cheers

robert-hh, 2017-04-20:
All of what I have seen so far from Pymakr is a trail of fail. It seems to create more problems than solutions. 
Many people seem after trying to go back to basic tools. And, b.t.w., the existing Pymakr works as good (ord bad) with Pyboard & co. than with Pycom devices.

mohpor, 2017-04-20:
@robert-hh 
This is not about the Pycom, even if they are failing at everything, the idea of a more robust coding environment seems tempting, don't you agree?

danicampora, 2017-04-20:
@robert-hh have you tried the Atom plug-in? I know that the full Pymakr IDE has lots of issues, but the Atom plug-in (although still in beta) works really well. We held a Hackathon in Shanghai last Monday with 40 participants, all of them using the plug-in and it worked like a charm. Syncing works great now.

robert-hh, 2017-04-20:
@danicampora Hello Daniel, how is life in China. I heard you have a good and interesting time there, improving your core products.
Yes, I tried the Atom plug-in, and after some hacking it worked. And if you look backinto the communication about Pymakr and the Atom Plug-in now, you know the problem. It is not whether you like Atiom or any other integrated development system. But when you offer such a thing, you earn besides the task to make a robust software itself all the trouble of all various PC configurations, the users may have. You will spend for ever a reasonable amount of you companies skill solving all that weird stuff, which is in no way supportive for your product and intentions. And that's a pity because you have already enough open leads to cover.

danicampora, 2017-04-20:
@robert-hh it's going well, China is an interesting place in many aspects :-)

> Yes, I tried the Atom plug-in, and after some hacking it worked.

True, it doesn't install at first try depending on the system configuration (which we are improving already), but once it's installed it's rock solid.

> You will spend for ever a reasonable amount of you companies skill solving all that weird stuff, which is in no way supportive for your product and intentions.

I agree that it consumes a good amount of resources, but for a lot of people having a proper IDE is a must.

prusnak, 2017-04-20:
Personally, I like VSCode more than Atom. But it might be possible to write a plugin that works both in Atom and VSCode.

pfalcon, 2017-08-23:
Closing due to inactivity. Please continue discussion on the forum: https://forum.micropython.org/

closed 2017-08-23 by pfalcon