micropython/micropython #3038: A perplex about the ">=" operator

YeWeiLiWeiwei, 2017-04-24:
I am a micropython novice.
CODE:
>>> intvar = int(1)
>>> intbytes = intvar.to_bytes(4, 'little')
>>> intvar_back = int.from_bytes(intbytes, 'little')
>>> intvar
1
>>> intvar_back
1
>>> type(intvar)
<class 'int'>
>>> type(intvar_back)
<class 'int'>
>>> intvar >= intvar_back
False

peterhinch, 2017-04-24:
I can't replicate this. Tested on Unix build
```
MicroPython v1.8.7-659-g5e66f2b on 2017-04-23; linux version
```
 What platform did you test on?

robert-hh, 2017-04-24:
I can replicate that on an esp8266, and also:
```
 >>> intvar < intvar_back
True
```

YeWeiLiWeiwei, 2017-04-24:
I tested on:
`MicroPython v1.8.7 on 2017-04-21; PYBv1.0 with STM32F405RG`
In this case, it is correct:
`(intvar - intvar_back) >= 0`
True

robert-hh, 2017-04-24:
This one is fine:
>>> (intvar * intvar_back) >= intvar_back
True
So I guess it is cause by int.from_bytes converting back to a different representation of int (long int), and thus the comparison fails, because long ints are by default considered larger than small ints.

pfalcon, 2017-04-24:
> because long ints are by default considered larger than small ints.

?? How can that be? "by default"

stinos, 2017-04-24:
>  a different representation of int (long int), and thus the comparison fails

That's not exactly what happens. Both ints get converted to mpz before comparision so both use the same representation. I'm not sure if the comparision is at fault here, or the mpz from_bytes implementation (which yields an mpz with len 2, that seems incorrect?).

YeWeiLiWeiwei, 2017-04-24:
I guess it is cause by int.from_bytes.(MicroPython v1.8.7 on 2017-04-21; PYBv1.0 with STM32F405RG
`>>> intvar = int(10)`
`>>> intbytes = intvar.to_bytes(4, 'little')`
`>>> intvar_back = int.from_bytes(intbytes, 'little')`
`>>> intvar -= 1`
`>>> intvar_back -= 1`
`>>> intvar >= intvar_back`
True
when "intvar_back -=1" is executed, it will construct a new int variable(int is constant).Then, comparision is ok.

peterhinch, 2017-04-24:
So it's platform dependent. I can't replicate under the Unix build, but can confirm on the Pyboard V1.1:
```python
MicroPython v1.8.7-659-g5e66f2b on 2017-04-23; PYBv1.1 with STM32F405RG
Type "help()" for more information.
# lines omitted 
>>> intvar_back == 1
False
>>> intvar_back
1
>>> isinstance(intvar_back, int)
True
>>> 
```

pfalcon, 2017-04-24:
> I can't replicate under the Unix build

Did you try 32-bit unix build?

robert-hh, 2017-04-24:
I CAN replicate the bug with a 32 bit unix

robert-hh, 2017-04-24:
I can also replicate the bug on 64 bit Unix, if I change the second line of the test to:
`intbytes = intvar.to_bytes(8, 'little')`

stinos, 2017-04-25:
I assume the platfrom dependency might be because depening on 32/64bit etc a small int might be used instead of mpz, and the former won't have the problem?

pfalcon, 2017-04-25:
There were seemingly related commits from @dpgeorge, let's wait for his comments.

robert-hh, 2017-04-25:
The latest commit of damien seems to have fixed the issue, as far as I could test it. The 32/64 bit difference has an impact on the size of an int that can be held in a single word. 
32 bit architectur <-> 30 bit int, problem occures with bytes of size 4 and up
64 bit architecture <-> 62 bit inr, problem occures with bytes of size 8 and up

dpgeorge, 2017-04-25:
Thanks @LiWeiwei-GitHub for the bug report.  There were a few issues similar to this before, where big ints were not made canonical and didn't compare correctly to small ints (or other big ints) of the same value.

In the case at hand, converting bytes to a big int where there were zero-valued bytes in the most significant bytes (eg `b\x01\x00\x00\x00\x00\x00\x00\x00\x00') lead to the creation of a big int that was not of canonical form.  This should be now fixed by c7aa86ce6f17f52aadff898b79c95d968da6bfa2, and tests were added in 810133d97d4391151e86af90508222c480f362b7

closed 2017-04-25 by dpgeorge