micropython/micropython #3041: Confused about _thread.allocate_lock().acquire() in main thread and worker threads

LiWeiwei-GitHub, 2017-04-25:
I recently used PYBV10 to write a project.

In main.py, i want process some data(main thread).The data come from other worker threads.In most time, the main thread should be stop state by lock.acquire().When the worker threads have data, wake up the main thread(lock.release()).

=====main.py=====
`...
LOCK = _thread.allocate_lock()
LOCK.acquire()

while True:
	...
	LOCK.acquire()
	...
...`
=================

Execute the above program, it will tell me "FATAL ERROR:deadlock".

When i move it to a worker thread, it works fine:

=====move to worker thread=====
`...
def worker_thread():
	...
	LOCK = _thread.allocate_lock()
	LOCK.acquire()
	
	while True:
		...
		LOCK.acquire()
		...
	...

_thread.start_new_thread(worker_thread, ())
...`
===============================

I`m a python starter, i can`t understand why the main thread can not be like the other worker threads.

dpgeorge, 2017-04-26:
It looks like you really do run into a deadlock in the first example.  In that case you have 1 thread (the main thread) and it's trying to acquire the lock a second time, without first releasing it.  In such a case the main thread will block forever (a deadlock) because it can't unlock its own lock, and there are no other threads to do it.

In the second case there is no deadlock because the main thread could unlock the lock that the worker thread has locked twice.

Maybe you want to try these things with normal Python first, which works in the same way.

Otherwise, this is not a bug so closing the issue.

closed 2017-04-26 by dpgeorge

dmazzella, 2017-04-26:
I had the same problem by only enabling MICROPY_PY_THREAD, try also to enable MICROPY_PY_THREAD_GIL

#define MICROPY_PY_THREAD           (1)
#define MICROPY_PY_THREAD_GIL       (1)

LiWeiwei-GitHub, 2017-04-26:
@dmazzella I enable MICROPY_PY_THREAD_GIL too. It does not work for me.

LiWeiwei-GitHub, 2017-04-26:
@dpgeorge I known the first example will block forever(a deadlock) because i (the main thread) acquire the lock twice.In normal Python, the first example will block forever, it will not show me deadlock message, just block there.That is what i want.In my program, the lock will be release by other worker thread.
`Python 3.6.0 (v3.6.0:41df79263a11, Dec 22 2016, 17:23:13) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import _thread
>>> lock = _thread.allocate_lock()
>>> lock.acquire()
True
>>> lock.acquire()
`