micropython/micropython #3042: Unsetting Makefile VFAT flag kills firmware.elf linking

cefn, 2017-04-25:
I'm trying to find firmware build configuration flags to disable unnecessary functions, as well as any other techniques for reclaiming RAM as described at https://forum.micropython.org/viewtopic.php?f=16&t=3312 

I want to use a NodeMCUv2 strictly as a standalone Micropython interpreter with GPIO/SPI support; debugging or network support are unused in our deployment. I hope to reduce power usage, preserve FlashCode memory space and ESPECIALLY make more RAM available.

One of the strategies is to toggle Makefile flags to see what RAM advantages I get. I'm not sure if there's any reference material for how to disable features properly, or what features are available to be suppressed to gain resources. Incidentally I would really be grateful for suggestions on this or if there is any reference material I should explore.

So far I successfully suppressed the use of MICROPY_PY_USSL, MICROPY_SSL_AXTLS and MICROPY_PY_BTREE within the Makefile by setting them to 0 and the build completed successfully.

These changes meant that I could then change _main.c_ to read...
```STATIC char heap[48 * 1024];```
...and the image ran without instability, gaining me critical RAM resources.

I have no idea whether this is legitimate, could break something else, or how to know what the magic number of increased available RAM is, given the features I have disabled in the build. However, without the corresponding Makefile changes, this increased heap led to crashes. No idea if there are other flags/values I should be tweaking to gain more RAM, or ensure that memory allocations line up.

During this process of blindly attempting to reclaim resources, I found that attempting to suppress MICROPY_FATFS causes a build error.

After setting the Makefile to...
```MICROPY_FATFS = 0```

Running...

```
make clean
make build/firmware-combined.bin
```

Gives a build which proceeds until the firmware.elf stage, which fails with the error shown below. My expectation that setting a Makefile flag to suppress library inclusion would cause dependent libraries not to be linked later. I am filing it as it seems to be an error somewhat like https://github.com/micropython/micropython/issues/2475 in that changes to downstream build instructions should probably have macros to handle when VFAT support is not intended to be included. 

I am hoping that there is a RAM benefit to not loading the VFAT libraries and filesystem, though not sure what changes I need to make to gain this benefit, aside from toggling MICROPY_FATFS. 

Suggestions as to other major build options I can tweak to create a Networking/Filesystem-free Python IO board with maximum runtime RAM would be gratefully received, even if they make code undebuggable.

```
LINK build/firmware.elf
build/py/../extmod/vfs_fat.o:(.text.vfs_fat_umount+0x4): undefined reference to `f_umount'
build/py/../extmod/vfs_fat.o: In function `vfs_fat_umount':
vfs_fat.c:(.text.vfs_fat_umount+0x11): undefined reference to `f_umount'
build/py/../extmod/vfs_fat.o:(.text.vfs_fat_mount+0x0): undefined reference to `f_mount'
build/py/../extmod/vfs_fat.o:(.text.vfs_fat_mount+0x4): undefined reference to `f_mkfs'
build/py/../extmod/vfs_fat.o: In function `vfs_fat_mount':
vfs_fat.c:(.text.vfs_fat_mount+0x2e): undefined reference to `f_mount'
vfs_fat.c:(.text.vfs_fat_mount+0x50): undefined reference to `f_mkfs'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_statvfs+0x0): undefined reference to `f_getfree'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_statvfs':
vfs_fat.c:(.text.fat_vfs_statvfs+0x15): undefined reference to `f_getfree'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_stat+0xc): undefined reference to `f_stat'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_stat':
vfs_fat.c:(.text.fat_vfs_stat+0x7f): undefined reference to `f_stat'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_getcwd+0x0): undefined reference to `f_getcwd'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_getcwd':
vfs_fat.c:(.text.fat_vfs_getcwd+0x19): undefined reference to `f_getcwd'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_chdir+0x0): undefined reference to `f_chdir'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_chdir':
vfs_fat.c:(.text.fat_vfs_chdir+0x19): undefined reference to `f_chdir'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_mkdir+0x0): undefined reference to `f_mkdir'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_mkdir':
vfs_fat.c:(.text.fat_vfs_mkdir+0x19): undefined reference to `f_mkdir'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_mkfs':
vfs_fat.c:(.text.fat_vfs_mkfs+0x2b): undefined reference to `f_mkfs'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_remove_internal+0x0): undefined reference to `f_unlink'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_remove_internal':
vfs_fat.c:(.text.fat_vfs_remove_internal+0x2a): undefined reference to `f_stat'
vfs_fat.c:(.text.fat_vfs_remove_internal+0x54): undefined reference to `f_unlink'
build/py/../extmod/vfs_fat.o:(.text.fat_vfs_rename+0x0): undefined reference to `f_rename'
build/py/../extmod/vfs_fat.o: In function `fat_vfs_rename':
vfs_fat.c:(.text.fat_vfs_rename+0x2e): undefined reference to `f_rename'
vfs_fat.c:(.text.fat_vfs_rename+0x46): undefined reference to `f_rename'
build/py/../extmod/vfs_fat_file.o:(.text.file_obj_close+0x0): undefined reference to `f_close'
build/py/../extmod/vfs_fat_file.o:(.text.file_obj_close+0xf): undefined reference to `f_close'
build/py/../extmod/vfs_fat_file.o:(.text.file_obj_write+0x0): undefined reference to `f_write'
build/py/../extmod/vfs_fat_file.o: In function `file_obj_write':
vfs_fat_file.c:(.text.file_obj_write+0x16): undefined reference to `f_write'
build/py/../extmod/vfs_fat_file.o:(.text.file_obj_read+0x0): undefined reference to `f_read'
build/py/../extmod/vfs_fat_file.o: In function `file_obj_read':
vfs_fat_file.c:(.text.file_obj_read+0x12): undefined reference to `f_read'
build/py/../extmod/vfs_fat_file.o:(.text.file_open+0x8): undefined reference to `f_open'
build/py/../extmod/vfs_fat_file.o:(.text.file_open+0xc): undefined reference to `f_lseek'
build/py/../extmod/vfs_fat_file.o: In function `file_open':
vfs_fat_file.c:(.text.file_open+0xc4): undefined reference to `f_open'
vfs_fat_file.c:(.text.file_open+0xf3): undefined reference to `f_lseek'
build/py/../extmod/vfs_fat_file.o:(.text.file_obj_ioctl+0x0): undefined reference to `f_sync'
build/py/../extmod/vfs_fat_file.o: In function `file_obj_ioctl':
vfs_fat_file.c:(.text.file_obj_ioctl+0x3a): undefined reference to `f_lseek'
vfs_fat_file.c:(.text.file_obj_ioctl+0x50): undefined reference to `f_sync'
build/py/../extmod/vfs_fat_misc.o:(.text.fat_vfs_listdir2+0xc): undefined reference to `f_opendir'
build/py/../extmod/vfs_fat_misc.o:(.text.fat_vfs_listdir2+0x10): undefined reference to `f_readdir'
build/py/../extmod/vfs_fat_misc.o:(.text.fat_vfs_listdir2+0x14): undefined reference to `f_closedir'
build/py/../extmod/vfs_fat_misc.o: In function `fat_vfs_listdir2':
vfs_fat_misc.c:(.text.fat_vfs_listdir2+0x41): undefined reference to `f_opendir'
vfs_fat_misc.c:(.text.fat_vfs_listdir2+0x73): undefined reference to `f_readdir'
vfs_fat_misc.c:(.text.fat_vfs_listdir2+0xf6): undefined reference to `f_closedir'
build/py/../extmod/vfs_fat_misc.o: In function `fat_vfs_import_stat':
vfs_fat_misc.c:(.text.fat_vfs_import_stat+0xe): undefined reference to `f_stat'
Makefile:196: recipe for target 'build/firmware.elf' failed
make: *** [build/firmware.elf] Error 1
```

pfalcon, 2017-04-25:
> These changes meant that I could then change main.c to read...
> STATIC char heap[48 * 1024];
> ...and the image ran without instability, gaining me critical RAM resources.

That's not true. MicroPython is already super-optimized, so disabling random features won't gain you *RAM*.

cefn, 2017-04-25:
No question Micropython is optimised, but I'd still expect there would be a relationship between disabling features which are on by default (and which would otherwise use RAM) and regaining the RAM. 

However, I don't think I've even begun to understand which features are in fact on by default, and how to disable them. A fairly concrete example is the VFat filesystem initialisation in the default _boot.py. Commenting this out seems to have gained me something, which is probably what I should have done in the first place.

I think I understand you're saying that the Makefile flags I'm setting don't change the behaviour of the board in any way that would be relevant to RAM, (they are maybe just changing which libraries are included into the Flash Code area). However, this is also very relevant given I am placing frozen modules there, so increasing available space is handy (I hope I've got the right notion of which memory area is which, sorry if I haven't).

Certainly seemed suspicious to me that the poster at https://gerfficient.com/2016/10/19/fix-to-micropythons-memory-allocation-problem-on-esp8266/ suggested just increasing heap in this way. However, it seems functional, and stability seemed to be dependent on disabling the features in the Makefile, though could have been a coincidence somehow. 

Having changed the heap number, I've tested allocating ```b=bytearray(x)``` where x is close to the limit of available RAM. I must be overwriting something or another and getting away with it by expanding the heap. Not sure what. On boot I currently seem to have quite a lot of RAM available with the various changes I've made, and all seems well...

```
>>> b=bytearray(47000); del b; gc.collect(); gc.mem_free()
47888
```

Not sure I can explain what it is that I've done, or at what cost.

However this is probably off-topic for the build flag error. Very happy to discuss any of these issues further in the dedicated thread about reclaiming RAM in my project at https://forum.micropython.org/viewtopic.php?f=16&t=3312 if it suits.

dpgeorge, 2017-04-26:
To focus on the actual issue of this ticket, the "problem" is that there are multiple config variables to enable/disable the VFS (and FatFS driver) on esp8266.  To disable it requires turning off MICROPY_FATFS and MICROPY_VFS and MICROPY_VFS_FAT all at the same time; doing only one will give link errors.  See esp8266/mpconfigport_512k.h for how it's disabled in the 512k target.

@pfalcon given that we now require gc-sections to work, we could actually remove the MICROPY_FATFS config option from the esp8266 makefile and just always compile and link the oofatfs/*.c sources, and exclusively use MICROPY_VFS and MICROPY_VFS_FAT at the level of the C preprocessor (this is how unix does it).  BUT I tried this and, while esp8266 512k target links correctly and doesn't include FatFS, esptool.py (incorrectly) gives a warning about missing symbols (namely disk_ioctl, disk_read, disk_write) in the final elf file.  So that's a bit of a pain.

PR #5304 (2019-11-11): extmod: Consolidate FAT FS config to MICROPY_VFS_FAT across all ports.
This commit removes the Makefile-level MICROPY_FATFS config and moves the MICROPY_VFS_FAT config to the Makefile level to replace it.  It also moves the include of the oofatfs source files in the build from each port to a central place in extmod/extmod.mk.
    
For a port to enabled VFS FAT support it should now set MICROPY_VFS_FAT=1 at the level of the Makefile.  This will include the relevant oofatfs files in the build and set MICROPY_VFS_FAT=1 at the C (preprocessor) level.

The main aim is to centralise and standardise  [... cut]
closed 2019-11-11 by dpgeorge