micropython/micropython #3043: unix: Change the way minimal/coverage/nanbox/freedos targets are defined in Makefile

dpgeorge, 2017-04-26:
Currently the unix build has convenience targets to build various versions (eg coverage, nanbox, freedos).  Building such a variant is done simply by, eg, `make coverage`.  Under the hood this calls make recursively with certain config options set to values different to the default values (eg CFLAGS_EXTRA is different).

Calling make recursively has various problems (eg https://github.com/micropython/micropython/pull/3036#issuecomment-297211914).  It also means you can't do things like `make coverage clean`, that doesn't work as expected.  And things like the axtls library are actually built into the standard build/ directory, not build-coverage/.

I propose to fix these issues by doing a similar thing done in stmhal: there is a "BOARD" variable which is used to set the variant of the build, and the config options in the makefile (eg CFLAGS_EXTRA) will be set depending on the value of "BOARD" (of course a different name could be used, like TARGET).  Then one could do things like:
```
$ make BOARD=coverage
$ make BOARD=coverage clean
$ make BOARD=coverage deplibs
$ make BOARD=coverage test # run standard tests, not coverage specific ones, using coverage build
```
For convenience, we could also provide a "coverage" target which would recursively call make with BOARD=coverage (and for the other targets, minimal, nanbox, freedos).

These changes could also go across to esp8266, for the ota and 512k builds (which have similar issues).

pfalcon, 2017-04-26:
Yeah, while I usually repeat "make works well" mantra to people to bring up issue of other build systems (unrelated to uPy), make does show its limitations, especially lately. You're aware, that for Zephyr, I switched "minimal" builds to a wrapper script, right?

Will need to think about the proposal here when have more free mind, but using "coverage" value for BOARD sounds pretty hacky (why I introduced wrapper script above).

stinos, 2017-04-26:
> make BOARD=coverage

Do you really plan to use it like this, or rather something more faithful to the language like ```make VARIANT=coverage```?

dpgeorge, 2017-04-26:
> but using "coverage" value for BOARD sounds pretty hacky
> ...
> Do you really plan to use it like this, or rather something more faithful to the language like make VARIANT=coverage?

I don't mind if it's BOARD or TARGET or VARIANT or something else.  I just used BOARD because that relates to stmhal and helps to explain the concept.  As I say above, we can provide convenience targets, eg:
```
ifeq (VARIANT,coverage)
CFLAGS_EXTRA += ...
endif
ifeq (VARIANT,minimal)
...
endif

minimal:
    $(MAKE) VARIANT=minimal

coverage:
    $(MAKE) VARIANT=coverage
```

> ... why I introduced wrapper script above

I'd rather not have wrapper scripts.  A single makefile is cleaner IMO, and stmhal shows that using something like BOARD can handle quite complicated configurations.

pfalcon, 2017-05-11:
Just for the record, cc3200 port uses BTARGET, BTYPE for selectors beyond BOARD, e.g.:

~~~
make BTARGET=bootloader BTYPE=release BOARD=LAUNCHXL
~~~

We unlikely will want to inherit that, so should target to change that too.

dpgeorge, 2020-01-12:
Implemented by bd2fff66875ed5adab8ce163a7f2bdafbd0332f9

closed 2020-01-12 by dpgeorge