micropython/micropython #3044: btree should be used as ContextManager?

cefn, 2017-04-26:
Given the note on use of btree in https://github.com/micropython/micropython/commit/bb296482c385d51f1eb8041e18beb5aa9668637c

```
# Don't forget to close the underlying stream!
```

Doesn't this suggest that the library should implement and document btree as a context manager, allowing use of the ```with``` statement... https://docs.python.org/3/reference/compound_stmts.html#the-with-statement 

If btree was a context manager, docs could guide users towards this less error-prone way of working? Or is the cost of open/close sufficiently high that in practice, applications should leave the db open rather than open, write, close for a set of data?

pfalcon, 2017-06-05:
Let's look at the situation: you submitted #3042 asking how to get smaller code size, and then almost immediately this ticket, asking for increase in the code size. Because any feature increases code size. We keep number of features under strict control, using the following principles: https://github.com/micropython/micropython/wiki/ContributorGuidelines . Based on that, the answer to the question in the title is: "No, btree should not be used as ContextManager. However, if you like to, you're more than welcome to implement that on top of it." (Similar logic of course applies to many other cases.)

closed 2017-06-05 by pfalcon

cefn, 2017-06-05:
I agree RAM size trades off against features, as discussed in both #3042 and this issue. Strictly-speaking issue #3042 related to a flag that broke the build when toggled. Sorry for taking the thread wildly off topic in the detail I provided.

I can see how subclassing would allow me to use the ContextManager design pattern in my own code, but I thought it might benefit others to have it upstream. Thanks for considering my suggestion that implementing ContextManager would be valuable enough to pay the price of extra RAM, and I absolutely accept that you disagree.  I am also probably clueless about how much extra RAM would be implied by this change, so you are in a much better position to judge.

pfalcon, 2017-06-05:
> but I thought it might benefit others to have it upstream

It's impossible to have all ideas upstream, and for implementation like MicroPython we should have upstream as little as possible (we already have a lot more than any other embedded language has, just by virtue being a Python). Nor that's how Python community works. If you think some feature will benefit other users, write a module for it, post it on PyPI, and promote/support/maintain it.