micropython/micropython #3046: webrepl fails with daily build on esp8266

dracode, 2017-04-27:
After flashing the daily build, connecting to it over serial and enabling webrepl by the recommended `import webrepl_setup` method, this is what I get

From the computer side:
```
pi@raspberrypi:~ $ webrepl_cli.py ws.py 192.168.0.134:
put 192.168.0..134 8266
ws.py -> /ws.py
Password: 
Remote WebREPL version: (1, 8, 7)
Traceback (most recent call last):
  File "/home/pi/bin/webrepl_cli.py", line 250, in <module>
    main()
  File "/home/pi/bin/webrepl_cli.py", line 244, in main
    put_file(ws, src_file, dst_file)
  File "/home/pi/bin/webrepl_cli.py", line 126, in put_file
    assert read_resp(ws) == 0
  File "/home/pi/bin/webrepl_cli.py", line 100, in read_resp
    data = ws.read(4)
  File "/home/pi/bin/webrepl_cli.py", line 63, in read
    assert len(hdr) == 2
AssertionError
```

From the ESP8266 side:
```
>>>
WebREPL connection from: ('192.168.0.58', 33038)
dupterm: Exception in read() method, deactivating: OSError: [Errno 19] ENODEV
```

Same error with both the 20160426 and 20160425 dailies.

On reverting to the `esp8266-20170108-v1.8.7.bin` version, webrepl again works as expected.

dracode, 2017-05-02:
I continue to get the same error with the 20170502 build as well.  Am I the only one experiencing this problem...?  Perhaps I'm doing something incorrect?

I turned on debugging for the webrepl client, in case this extra info is useful:
```
pi@raspberrypi:~ $ webrepl_cli.py ws.py 192.168.0.134:
put 192.168.0.220 8266
ws.py -> /ws.py
'WA\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' 82
Got unexpected websocket record of type 81, skipping it
Skip data: 
WebREPL connected
>>> 
Remote WebREPL version: (1, 8, 7)
'WA\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00c\x14\x00\x00\x06\x00/ws.py\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' 82
Traceback (most recent call last):
  File "/home/pi/bin/webrepl_cli.py", line 250, in <module>
    main()
  File "/home/pi/bin/webrepl_cli.py", line 244, in main
    put_file(ws, src_file, dst_file)
  File "/home/pi/bin/webrepl_cli.py", line 126, in put_file
    assert read_resp(ws) == 0
  File "/home/pi/bin/webrepl_cli.py", line 100, in read_resp
    data = ws.read(4)
  File "/home/pi/bin/webrepl_cli.py", line 63, in read
    assert len(hdr) == 2
AssertionError
```

pfalcon, 2017-05-02:
This looks related to https://github.com/micropython/micropython/issues/2850 and https://github.com/micropython/micropython/pull/2950 .

pfalcon, 2017-05-06:
#2850 was resolved, so this should work now. Closing, but feel free to reopen if needed.

closed 2017-05-06 by pfalcon