micropython/micropython #3054: .mpy files repeat source filename for each function stored, or something

pfalcon, 2017-04-30:
There's something wrong with .mpy files: they store the string value of source filename again and again, multiple times:

~~~
./lib/time.py.ustruct.unpack
@iiiiiiiii.tuple...tmg.............&G(0((+......7......d.......d.............Q.f.......d...h...d..........d.........X.!...d.[.strftime
./lib/time.py.time.int.array.array.i
localtime_      bytearray       strftime_.str.utf-8...format.tR............)&G(.(......7......d.......d.............f.......d...
...........f.d.[        localtime
./lib/time.py.time.int.ustruct.pack.i
localtime_._c_tm_to_tuple.uctypes.bytearray_at...tR............3&G(.(......7......d.......d.............f.......d...
...........f.d.[.gmtime
./lib/time.py.time.int.ustruct.pack.i.gmtime_._c_tm_to_tuple.uctypes.bytearray_at...t.......            ....=........
...d.d.[.mktime
./lib/time.py.mktime_._tuple_to_c_tm...tt.......        .....A.......d.[.perf_counter
./lib/time.py.time.........     .....D.......d.[.process_time
./lib/time.py.clock..
~~~

That certainly doesn't help to save heap size when loading them.

[time.zip](https://github.com/micropython/micropython/files/967001/time.zip)

pfalcon, 2017-04-30:
Well, this actually applies not just to src filename, but to other strings (apparent qstr's!) too.

dpgeorge, 2017-04-30:
> There's something wrong with .mpy files: they store the string value [...] again and again

That's by design.  To keep the .mpy saver and loader small it doesn't keep a table of previously stored qstrs, so must save each one each time it encounters it.

> That certainly doesn't help to save heap size when loading them.

It doesn't affect heap size (only on-disk size).  Each unique qstr is only loaded once, subsequent ones are found in RAM and their index reused (by virtue of it being a unique-str).

pfalcon, 2017-05-01:
Well, it's still a little bit confusing, at least while we don't have good tools to assess a memory map.

pfalcon, 2017-05-11:
As discussion in https://github.com/micropython/micropython/issues/2845#issuecomment-300640119 reminds again, the size matters (it talks about frozen bytecode which isn't (not as much) susceptible to the problem in this ticket, but still).

One of the promise of using scripting languages is that by using high-level bytecode, one can get much higher code density than with machine code. Well, currently, we have situation in uPy when bytecode size is almost the same as source size. Yeah, that delivers seamless and deep dynamic linking, beautiful and clear error reports, etc. - but, it doesn't deliver on the code density side.

So, we should look into trying to optimize it. This shouldn't go on its own, but as part of other changes, e.g. change to allow to *map* .mpy files, instead of *loading* them.

And not a priority for now at all, we should finish with current sagas before starting on new exciting things ;-).

pfalcon, 2018-12-23:
So, the current implementation is indeed optimized for memory used during .mpy loading vs .mpy size, or loading time overheads.

Resolving that would require some temporary memory during loading. The idea would be to have common qstr pool for the entire module, i.e. .mpy file. The .mpy would then start with this pool of qstrs, and loader would allocate `uint16_t qstrs[num]`, and store id of each interned qstr there. Then stored bytecode would contain index of an entry in this per-module table, and during loading, it would be patched with the actual id from the table. The table is not needed once module is loaded.

Current mpy-cross implementation can be refactored to use new format, but still save duplicate strings in the pool, to keep saving algo simple. Really, all that optimization stuff should be written in Python (MicroPython!), not C.

This same idea might apply to other .mpy pools, e.g. object pool - it rather should be per-module, which offers possibility to avoid storing duplicate entries in per-function pools otherwise.

If there's a concern that a single pool may be too large, the could provide support for multiple pools, each potentially reused by more than one code object.

dpgeorge, 2019-02-22:
I did some analysis on the possible savings that could be made by eliminating the duplicate qstrs in .mpy files.  Some results:

| file | original size | reduced size | fraction | N unique qstr | N unique qstr per 1024 bytes |
| --- |---|---|---|---|---|
| `lcd160cr.mpy` | 9902 | 7303 | 0.74 | 152 | 21.31
| `lcd160cr_test.mpy` | 3144 | 2597 | 0.83 | 61 | 24.05 |  
| `onewire.mpy` | 1711 | 1263 | 0.73 | 27 | 21.89 |  
| `dht.mpy` | 968 | 683 | 0.70 | 12 | 17.99 |  
| `upip.mpy` | 5343 | 4434 | 0.82 | 95 | 21.93 |  
| `upip_utarfile.mpy` | 2080 | 1426 | 0.68 | 33 | 23.69 |  
| `websocket_helper.mpy` | 1183 | 1016 | 0.85 | 18 | 18.14 |  
| `webrepl.mpy` | 1912 | 1396 | 0.73 | 36 | 26.40 |  
| `ntptime.mpy` | 661 | 553 | 0.83 | 22 | 40.73 |  
| `urequests.mpy` | 2260 | 1848 | 0.81 | 44 | 24.38 |  
| `uasyncio/__init__.mpy` | 5471 | 4083 | 0.74 | 86 | 21.56 |  
| `uasyncio/core.mpy` | 6484 | 4561 | 0.70 | 79 | 17.73 |  
| `uasyncio/queues.mpy` | 1415 | 977 | 0.69 | 21 | 22.01 |  
| `umqtt/robust.mpy` | 945 | 734 | 0.77 | 21 | 29.29 |  
| `umqtt/simple.mpy` | 3700 | 2781 | 0.75 | 51 | 18.77 |   

This data is obtained with #4473 (a set of static qstrs) already applied to see what the ultimate savings could be.  The total .mpy size with duplicate qstrs (sum of second column) is 47179 bytes.  The total .mpy size after duplicates are removed (sum of third column) is 35655 bytes.  This is a saving of about 25%, which is about the average saving that could be expected from eliminating duplicate qstrs.

The "N unique qstr" column counts how many unique qstrs there are in the .mpy file, and show how much memory would be needed for the "qstr symbol table" if it were all loaded into RAM.

The final column is a normalised version of the former column, to see what the average number of unique qstrs are per kbyte of .mpy file.  The average is about 25 unique qstrs per 1024 bytes of .mpy.  In terms of memory usage, this would be 50 bytes of RAM for each 1024 bytes of .mpy, or about 5%.

So a broad summary would be: eliminating duplicate qstrs in .mpy files would save about 25% of storage space at the expense of about 5% extra RAM usage during load time.

-----

Having a global "qstr symbol table" sounds simple but actually makes saving (or loading) more complicated.  For saving it requires doing a pass over the data first to pull out the unique qstrs, then saving them to the .mpy file at the start of the file.  Then passing over the data a second time to save the bytecode.

Instead of this it would be possible to store the "qstr symbol table" incrementally, only as they are needed.  This would enable the save and load to happen in a single pass.  Eg, when a qstr is needed, if it has not been seen before then write the data, otherwise write a reference to a previous qstr.

But as the analysis above shows, the number of unique qstrs is still rather large.  What I would propose is to do something similar to dictionary compression and have a fixed-size qstr symbol table that is a sliding window of the previous N qstrs.  As new qstrs are encountered in the .mpy file they are pushed onto this FIFO table.  If a previous qstr already exists in the FIFO then it is referenced by index into the FIFO.

Choosing a value for N of around 32 would allow the FIFO to be stored on the C stack, and so eliminate any need for heap allocation and associated fragmentation.  And for most small-to-medium .mpy files N=32 might be enough to hold all unique qstrs (see above table, N unique qstr column), so this would be just as good as the simple qstr table mechanism.  For very large .mpy files it would mean that the RAM usage for the qstr symbol table is bounded.

dpgeorge, 2019-02-25:
> What I would propose is to do something similar to dictionary compression and have a fixed-size qstr symbol table that is a sliding window of the previous N qstrs.

I implemented this idea in #4549.  The results are (for window size N=32):

| file | original size | reduced size | fraction
-|-|-|-
`lcd160cr.mpy` | 10439 | 8360 | 0.80
`lcd160cr_test.mpy` | 3266 | 2801 | 0.86
`onewire.mpy` | 1940 | 1368 | 0.71
`dht.mpy` | 1128 | 755 | 0.67
`upip.mpy` | 5907 | 5197 | 0.88
`upip_utarfile.mpy` | 2439 | 1679 | 0.69
`websocket_helper.mpy` | 1304 | 1088 | 0.83
`webrepl.mpy` | 1748 | 1275 | 0.73
`ntptime.mpy` | 684 | 576 | 0.84
`urequests.mpy` | 2609 | 2113 | 0.81
`uasyncio/__init__.mpy` | 6139 | 4404 | 0.72
`uasyncio/core.mpy` | 7796 | 5438 | 0.70
`uasyncio/queues.mpy` | 1730 | 1059 | 0.61
`umqtt/robust.mpy` | 1097 | 803 | 0.73
`umqtt/simple.mpy` | 4333 | 3169 | 0.73

dpgeorge, 2019-05-31:
Storage of qstrs in .mpy files was optimised in 5996eeb48fa6ce773003eacbc7e0b52ad5cc4d31 and 4f0931b21f72be86aae22f05b718aa36792afc9b

closed 2019-05-31 by dpgeorge