micropython/micropython #3061: Maybe-bug in esp8266/modnetwork.c (beginner in C, not sure)

Jostikas, 2017-05-03:
In esp8266/modnetwork.c in function esp_connect isn't there a leak when an unsecure network is joined?

     if (n_args > 1) {
        p = mp_obj_str_get_data(args[1], &len);  # len gets a presumably nonzero value
        memcpy(config.ssid, p, len);
        if (n_args > 2) {
            p = mp_obj_str_get_data(args[2], &len);
        } else {
            p = "";   # len still has length of ssid
        }
        memcpy(config.password, p, len); # Doesn't this copy more than the empty string when unsecured network is joined?

pfalcon, 2017-05-03:
Yes, that's non-fatal, but definitely a thinko. It was contributed via pull request. And that's the reason why we have 87 unmerged pull requests - because they need very thorough review, and testing. The maintainers can't keep up with that. The users interested in a patch need to get together and review and test it. Otherwise hasty merges lead to issues like that (in this case I'm too blame).

@puuu : FYI.

dpgeorge, 2017-05-06:
There's also potential buffer overflows here, don't know if we should fix that or not.

Jostikas, 2017-05-06:
As in, the ´len´-s gotten from the python string can be larger than 32 and 64 bytes for ssid and pw? Should be an easy check, no?

pfalcon, 2017-05-06:
@dpgeorge : My plan is to look at the API reference and update the code. This should not block 1.9 though.

puuu, 2017-05-12:
@jostikas thank you for pointing out the problem.

@pfalcon sorry for the inconvenience. This was definitely my fault. 

I created pull request #3083 that fix the discussed problems.

dpgeorge, 2017-06-01:
Fixed.

closed 2017-06-01 by dpgeorge