micropython/micropython #3072: ujson: Partial parsing support to reduce memory usage

pfalcon, 2017-05-06:
To further optimize upip (and any other application which processes large JSON files), it would be nice to have an ability to skip sections of input JSON stream, instead of having them parsed into data objects.
This can be achieved by adding a callback argument to .load(), which would be called apparently with a JSON dictionary key to be parsed and enclosing level of that key. The callback would return:

* Value1 - don't parse this element and all of its sub-elements.
* Value2 - parse this element, but continue calling callback for all of its subelements.
* Value3 - parse this element and all of its subelements, without calling callback on each subelement. The callback will be called on the next structural sibling of the current element.
* Value4 - terminate parsing.

The fact that it will be called only for dictionary elements means however that such optimization won't help with sequence JSON elements (so, the above scheme might be updated for that, or not).

pfalcon, 2017-05-06:
Another way to deal with that would be to add token stream based method parsing (SAX in Java terms), but with support to for structural parsing of any substream (DOM in Java terms). So, token stream parsing can be used to efficiently navigate to needed location(s) in the input stream, and then switched back to structural parsing for easy extraction of needed data.

dpgeorge, 2020-06-09:
It would be good to be able to parse only parts of a JSON input, to reduce the need for a large amount of RAM.  But there hasn't been any progress or interest on this issue so I'll close it for now.

closed 2020-06-09 by dpgeorge

patrick-rt, 2021-01-12:
Is there any progress at this topic?