micropython/micropython #3073: uselect: Standardizing POLLHUP behavior for bare-metal targets

pfalcon, 2017-05-06:
Offshot of https://github.com/micropython/micropython/issues/2931, see https://github.com/micropython/micropython/issues/2931#issuecomment-298159068

pfalcon, 2017-05-08:
Another interesting example (on Linux):

~~~
import usocket as socket
import uselect as select
import time

s = socket.socket()
a = socket.getaddrinfo("localhost", 80)[0][-1]
s.connect(a)

poller = select.poll()
poller.register(s, select.POLLIN)
res = poller.poll(500)
print(res)

s.close()

res = poller.poll(500)
print(res)
~~~

gives:

~~~
()
[(<_socket 3>, 32)]
~~~

This may be potentially useful to handle (in abrupt way) socket timeout with uasyncio.

pfalcon, 2017-05-08:
32 is POLLNVAL

dpgeorge, 2017-05-09:
The above example already works the same way on esp8266 (it returns "1" instead of "32", but "1" corresponds to POLLIN).

dpgeorge, 2020-06-09:
It looks like this was fixed by feaa2516742ed4f9032a0233dcec922ac6a4e80c

closed 2020-06-09 by dpgeorge