micropython/micropython #3076: ESP8266: ticks_ms() advances unexpectedly

robert-hh, 2017-05-07:
When taking tick_ms() in the presence of a pin interrupt, the value returned by ticks_ms() takes unexpected leaps. Simple test code below. I triggered the device with 100 Hz, 10 µs pulse width.
With the pulse off there is no problem.
```
from time import ticks_ms, ticks_diff, sleep_ms
from machine import Pin

# just a dummy handler
def sample(pin):
    pass

def main():
    last = ticks_ms()
    irq_pin = Pin(5, Pin.IN, Pin.PULL_UP)
    irq_pin.irq(handler=sample, trigger=Pin.IRQ_FALLING, hard=True)
    try:
        while True:
            tick = ticks_ms();
            diff = ticks_diff(tick, last)
            if diff > 35:
                print("unexpected value", diff)
            last = tick
            sleep_ms(32)

    except KeyboardInterrupt:
        irq_pin.irq(handler=None)

main()
```
The leap size is always 4295000 dec (418958 hex) or 1 less. 
If you start the test, it may run for a while without problems. But the problem will show up latest after a minute. Slight variations of the pulse frequency may cause more frequent events.

robert-hh, 2017-05-08:
Update:
The effect does NOT happen when:
a) the IRQ handler is set to soft (hard = false)
b) ticks_us() is used instead of tick_ms() 
So it looks as if there happen unexpected increases of system_time_high_word in esp_mphal.c

robert-hh, 2017-05-08:
Update 3: Changing lines 123..127 of ets_mphal.c into:
```
    // handle overflow of system microsecond counter
    ets_intr_lock();
    uint32_t system_time_cur = system_get_time();
    if (system_time_cur < system_time_prev) {
        system_time_high_word += 1; // record overflow of low 32-bits
    }
    system_time_prev = system_time_cur;
    ets_intr_unlock();
```
seems to cure the problem. But I'm not sure if that is the proper change.

dpgeorge, 2017-05-10:
> But I'm not sure if that is the proper change.

That is one way to fix it, but not the only one.

As you point out, that section of code that handles overflow is critical code and if an interrupt occurs between reading `system_get_time()` and updating `system_time_prev`, and that interrupt runs the `ets_loop_iter()` function (ie re-enters it) then there will be a double increase of `system_time_high_word`.

So the alternative way to fix it is to make sure that `ets_loop_iter` is never re-entered, ie it's disabled when an IRQ is executing.

@pfalcon what do you think, is `ets_loop_iter` supposed to be re-entrant?

pfalcon, 2017-05-10:
@robert-hh :

> Changing lines 122..127 of ets_mphal.c into:

What these lines had before? Why don't you post diff instead?

pfalcon, 2017-05-10:
@dpgeorge 

> @pfalcon what do you think, is ets_loop_iter supposed to be re-entrant?

As a generic answer, the whole idea of our implementation is that we re-enter ets_main_loop(), which is otherwise noreturn function in the BootROM (in the later SDK versions Espressif themselves dupped it out of mask-ROM to adjust behavior). I can't say right away whether it should be re-entrant from an IRQ, I just remember that it does have some interrupt locking (not for the duration of the entire run), and last time I looked at the BootROM code again, it does a bit more than my impl of ets_loop_iter(), and yet it kinda works ;-). Bottom line: only experimentation may tell what will work and what not.

At the same time, the code posted by @robert-hh seems to be correct: if you perform multi-word operation, it needs to be done in a critical section, instead of trying to reason about the context of the "world" outside it.

robert-hh, 2017-05-10:
@pfalcon : 

> What these lines had before? Why don't you post diff instead?

Sorry for the confusion. The bracket around the critical section consisting of **ets_intr_lock();** and **ets_intr_ulock();** was added.

dpgeorge, 2017-05-11:
> As a generic answer, the whole idea of our implementation is that we re-enter ets_main_loop(),
> ...
> I can't say right away whether it should be re-entrant from an IRQ

Well it has most of the code protected via an ets_intr_lock/ets_intr_unlock, so the event hanlder loop is probably safe from IRQs.  And as such, copying this pattern of ets_intr_lock/ets_intr_unlock to the system time update makes sense (and seems to work as tested by @robert-hh).  So I'll go for that if there are no other ideas.

pfalcon, 2017-05-11:
Well, we probably should ask @robert-hh to submit his change as a pull request.

robert-hh, 2017-05-11:
That's what I was frightened about, because I'm not good at git. But I can give it another try. 
The way I had got some confidence in the change: w/o the change the error turned up typically within 30 seconds. With the change I had it running over the night w/o any error. As a proof good enough for engineers. Mathematicians or philosophers could object.

pfalcon, 2017-05-11:
@robert-hh : If it's a problem for you, @dpgeorge can do it, as he wrote, no worries.

robert-hh, 2017-05-12:
I gave the pull request a try.

dpgeorge, 2017-05-13:
Thanks for the fix @robert-hh.  For the record here is the test code I used which is standalone, and just requires to link pin 4 and pin 5:
```python
from time import ticks_ms, ticks_diff, sleep_ms
from machine import Pin, PWM

def sample(pin):
    for _ in range(10):
        pass

def main():
    last = ticks_ms()
    irq_pin = Pin(4, Pin.IN, Pin.PULL_UP)
    pulse = PWM(Pin(5), freq=500, duty=512)
    irq_pin.irq(handler=sample, trigger=Pin.IRQ_FALLING, hard=True)
    try:
        while True:
            tick = ticks_ms();
            diff = ticks_diff(tick, last)
            if diff > 35:
                print("unexpected value", diff)
            last = tick
            sleep_ms(32)

    except KeyboardInterrupt:
        irq_pin.irq(handler=None)

main()
```

closed 2017-05-13 by dpgeorge