micropython/micropython #3077: esp8266: neopixel crash

deshipu, 2017-05-10:
The following code consistently crashes my board:

```
MicroPython v1.8.7-292-g3d739eb on 2017-02-18; ESP module with ESP8266
Type "help()" for more information.
>>> import neopixel
>>> n = neopixel.NeoPixel(Pin(16), 2, False)
>>> n.write()
Fatal exception 28(LoadProhibitedCause):
epc1=0x40100e57, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 32120, room 16 
tail 8
chksum 0xca
load 0x3ffe8000, len 1088, room 0 
tail 0
chksum 0x04
load 0x3ffe8440, len 3000, room 8 
tail 0
chksum 0xe0
csum 0xe0
l��|��rrnb��l�b�lb쌜���lb�lrlll��|��rrnb��ll��b�b쌜��b��lrl�l��|��rrnb��l��b�b쌜��lb�lbl����n�r��n|�llll`��r�l�l�l`��r�l�l�l`��r�l���ll`rl��rl���b��b�bbr�rb��n�nn�l��l�l��ll������l�n����bll�rp���bl�brlrlr�n�����n����bpb������r�쎜�lbl��ln�prl�l�#4 ets_task(40100164, 3, 3fff8288, 4)
```
It seems that passing ``False`` as the ``is800khz`` makes it crash.

dpgeorge, 2017-05-11:
> It seems that passing False as the is800khz makes it crash.

The constructor's 3rd arg is bytes-per-pixel, and passing 0 is not a good idea :)  To get a 400kHz bitstream you need to use the low-level neopixel.neopixel_write function directly.

deshipu, 2017-05-11:
Then the docs at http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#neopixel-driver lie.

deshipu, 2017-05-11:
I see, I read the documentation wrong. Sorry for the confusion.

closed 2017-05-11 by deshipu