micropython/micropython #3086: open: No support for creating file if doesn't exist, while opening in r/w mode

pfalcon, 2017-05-13:
There doesn't seem to be a mode for `open()` to do subj. The closest one is "a+", but that additionally seeks to the end of file. Now the question is whether "a+" has a relation to POSIX open() mode O_APPEND which stipulates that any write to file opened with it happen at EOF (lseek() for writing is ignored).

Usecase: e.g., opening a file for btree.open().

pfalcon, 2017-05-13:
CPython3 docs https://docs.python.org/3/library/functions.html#open :

> 'a' for appending (which on some Unix systems, means that all writes append to the end of the file regardless of the current seek position).

man open:

~~~
       O_APPEND
              The  file  is  opened in append mode.  Before each write(2), the
              file offset is positioned at the end of the  file,  as  if  with
              lseek(2).
~~~

pfalcon, 2017-05-13:
fopen http://pubs.opengroup.org/onlinepubs/009695399/functions/fopen.html :

> Opening a file with append mode (a as the first character in the mode argument) shall cause all subsequent writes to the file to be forced to the then current end-of-file, regardless of intervening calls to fseek().

pfalcon, 2017-05-13:
Consequently, choices:

1. Accept the fact that Python can't create-or-open-without-truncation in a single op, without race conditions.
2. Unlike CPython, define explicitly that "a+b" allows seeks for writing, but such definition will directly contradict the same ANSI C mode string and indirectly well-known semantics of POSIX.
3. Define our own opening mode for this operation.

dpgeorge, 2017-05-14:
Option 4 is to go upstream and see if they are open to adding a new mode.

I don't like option 2.  I'd weakly prefer 1 over 3.

WitchDoc42, 2017-05-20:
I'm not sure i understand this discussion, open for write (mode w) creates the file if it doesn't exist? The POSIX standard linked to by pfalcon even says so;

> w or wb
>        Truncate to zero length **or create file** for writing.

dpgeorge, 2017-05-21:
The idea is that opening an existing file shouldn't truncate it.  Ie, open for read+write, creating if it doesn't exist, otherwise (if it does exist) leave it alone and seek to the start.  This is accomplished at the C-level by the flags `O_RDWR | O_CREAT` which isn't possible through Python.