micropython/micropython #3087: Need standard way to get string timestamp across ports

pfalcon, 2017-05-13:
Timestamps are pretty basic feature of real-world applications. In Python, they're achievable by time.strftime(). That's implemented for Unix port, but not baremetal ports. Baremetal ports have notion of real time only in machine.RTC class, but that's not natively available in Unix port. time.stftime() is convenient and flexible, but full implementation is kind of heavyweight (well, only kinda). From machine.RTC, there's kinda long way to get a timestamp (again, only kinda).

tomlogic, 2017-05-16:
Here's portable Python code we've been using to generate an ISO8601 timestamp using `time.localtime()`.  You can pass a timestamp to it, or it will use the current time as a default.

```
def timestamp(t = None):
	return "%04u-%02u-%02uT%02u:%02u:%02u" % time.localtime(t)[0:6]
```

pfalcon, 2017-05-31:
Kind of sounds good. When I submitted this ticket, I implicitly had in mind "how to get timestamp with minimal memory traffic". Getting a tuple and slicing it aren't exactly in that vein, but it's true it's kind of a solution. Now we just need to be sure that every port had utime.localtime(), which is achievable, as implementation is available in the core.

tomlogic, 2017-05-31:
The slice is only required because the string format ignores the day of week and day of year elements returned by `localtime()`.  Maybe there's another way to have the string formatter ignore those elements and not throw a TypeError of "not all arguments converted during string formatting".

MrSurly, 2017-05-31:
> Maybe there's another way to have the string formatter ignore those elements and not throw a TypeError of "not all arguments converted during string formatting".

`"{:04d}-{:02d}-{:02d}T{:02d}:{:02d}:{:02d}".format(*time.localtime())`

pfalcon, 2017-06-11:
unix port doesn't have utime.localtime()

pfalcon, 2017-06-11:
Version with % formatting takes twice less memory.

pfalcon, 2017-06-11:
Implemented.

https://github.com/micropython/micropython/commit/d42b80fd648d74e2c6f66626e904103091ee9334

https://github.com/pfalcon/micropython-btreedb/commit/4cc32a1ac9356ef402fc16ceaafc0191c5aee3e1

closed 2017-06-11 by pfalcon