micropython/micropython #3088: ESP8266: Spurious NameError in native code ISR

robert-hh, 2017-05-14:
When decorating an ISR with the micropython@native or micropython@viper decorator, spurious NameError message show up when the ISR interrupts another embedded modules operation. I have an example where a burst of SPI transfers is interrupted. These errors do not occurs if teh ISR is normal bytecode. The example consists of two files, amain file and the SPI using module. Integrating them into one let the error not show up. However I have an much lager code, where it does not matter whether the SPI module is separate or not. I have some faint reminder that this phenomenon was already mentioned either in the micropython forum or here, but could not find it. To be fair, the issue is not really dramatic, since it does not occur with bytecode ISR's. So it can be avoided, but still may be interesting.
Main code:
```
from time import sleep_ms, ticks_ms
from machine import Pin, PWM, SPI, freq
from array import array
from disp_f import SH1106_F

_BUFFERSIZE = const(64) # Size of the ringbuffer
#
_data = array("h", 0 for _ in range(_BUFFERSIZE))
_time = array("L", 0 for _ in range(_BUFFERSIZE))
_index_put = 0
#
# a not-so-fast dummy ISR

@micropython.native
def sample_n(pin, data = _data, time = _time):
    global _index_put
    for _ in  range (10):
        data[_index_put] = _index_put # data capture dummy
        time[_index_put] = ticks_ms()
        index_put = (_index_put + 1) % _BUFFERSIZE

def sample_b(pin, data = _data, time = _time):
    global _index_put
    for _ in  range (10):
        data[_index_put] = _index_put # data capture dummy
        time[_index_put] = ticks_ms()
        index_put = (_index_put + 1) % _BUFFERSIZE

def main(isr_native = True):

    spi = SPI(1, baudrate=10000000)

    display = SH1106_F(128, 64, spi, Pin(16), Pin(2))

    irq_pin = Pin(4, Pin.IN, Pin.PULL_UP)
    pulse = PWM(Pin(5), freq=250, duty=512)

    if isr_native:
        irq_pin.irq(handler=sample_n, trigger=Pin.IRQ_FALLING, hard=True)
    else:
        irq_pin.irq(handler=sample_b, trigger=Pin.IRQ_FALLING, hard=True)
    try:
        while True:

            display.show()

            sleep_ms(1000)
            print("------------")

    except KeyboardInterrupt:
        irq_pin.irq(handler=None)

main()
```
And imported module for SPI:
```
# a few register definitions
LOW_COLUMN_ADDRESS  = const(0x00)
HIGH_COLUMN_ADDRESS = const(0x10)
SET_PAGE_ADDRESS    = const(0xB0)

class SH1106_F:
    def __init__(self, width, height, spi, dc, res, external_vcc=False):
        self.rate = 10 * 1000 * 1000
        dc.init(dc.OUT, value=0)
        res.init(res.OUT, value=0)
        self.spi = spi
        self.dc = dc
        self.width = width
        self.height = height
        self.external_vcc = external_vcc
        self.pages = self.height // 8
        self.buffer = bytearray(self.pages * self.width)

    def show(self):
        mv = memoryview(self.buffer)
        for page in range(0, self.width * self.pages, self.width):
            self.dc(0)
            self.spi.write(bytearray([SET_PAGE_ADDRESS | page, LOW_COLUMN_ADDRESS | 2, HIGH_COLUMN_ADDRESS | 0]))
            self.dc(1)
            self.spi.write(mv[page:page + self.width])
```
For the test, GPIO4 and GPIO5 must be joined, a SPI device does not have to be connected. The test may work with a timer interrupt too. Version and build:
(sysname='esp8266', nodename='esp8266', release='2.0.0(5a875ba)', version='v1.8.7-751-g0c57979-dirty on 2017-05-13', machine='ESP module with ESP8266')

dpgeorge, 2017-05-14:
Looks very similar to #1573.

dpgeorge, 2017-06-07:
Closing as duplicate, see above-linked issue.

closed 2017-06-07 by dpgeorge