micropython/micropython #3089: ESP8266 socket.IPPROTO_UDP

pdxjohnny, 2017-05-14:
As shown in https://docs.micropython.org/en/latest/esp8266/library/usocket.html#usocket.socket.IPPROTO_UDP

there should be a IPPROTO_UDP defined in socket. but there is not, I am using today's build (esp8266-20170514-v1.8.7-751-g0c57979.bin)

Just wondering why that is an if i can help fix that somehow

pfalcon, 2017-05-14:
> Just wondering why that is an if i can help fix that somehow

Yup, just please read the documentation in full and spread the word: https://docs.micropython.org/en/latest/esp8266/library/index.html

> Note about the availability of modules and their contents: This documentation in general aspires to describe all modules and functions/classes which are implemented in MicroPython. However, MicroPython is highly configurable, and each port to a particular board/embedded system makes available only a subset of MicroPython libraries. For officially supported ports, there is an effort to either filter out non-applicable items, or mark individual descriptions with “Availability:” clauses describing which ports provide a given feature. With that in mind, please still be warned that some functions/classes in a module (or even the entire module) described in this documentation may be unavailable in a particular build of MicroPython on a particular board.

pdxjohnny, 2017-05-14:
ok cool, thanks!

closed 2017-05-14 by pfalcon