micropython/micropython #3092: AttributeError: 'Pin' object has no attribute 'id'

PikWay, 2017-05-15:
I'm trying to get pin number in irq callback handler, and getting above error. 
Found in doc that this attrib should be avail 

TC: 
from machine import Pin
SW1 = Pin(5, Pin.IN)  
SW1.id() 

returns exception:
AttributeError: 'Pin' object has no attribute 'id'

PikWay, 2017-05-15:
@esp8266 esp8266-20170108-v1.8.7.bin

dpgeorge, 2017-05-16:
The Pin.id() is an extension only provided by some ports, and not by esp8266.  Please see #2612 for previous discussion and other ways to get the id.

The real issue here is that the docs need updating to either remove this method (from the docs) or state that it's only available on some ports.

pfalcon, 2017-05-16:
Removing them on TODO. Waiting for final specs of Pin class (https://github.com/micropython/micropython/pull/3025) to clean up docs in one go (also giving a chance to interested parties to move, not remove superfluous methods somewhere else; based on docs change history, I was the only party interested in that, but I'm actually not, I agree with everyone that removing the corresponding code from the affected port is the better approach - remove non-standard stuff and find room to add something standard which was missing).

pfalcon, 2017-05-30:
Docs are updated: http://docs.micropython.org/en/latest/esp8266/library/machine.Pin.html .

Returning to the original question, it's similar to the following: I add numbers 3 and 7, and get sum of 10. Given this sum of 10, how do I get out of it numbers 3 and 7? Answer: there's no way. If you need these numbers after you sum, keep them around yourself - in variables or whatever. The same applies to a Pin - if you pass anything to a constructor and need it afterwards, keep it around. If you need to associate Pin object(s) with some additional data, be it "id" or anything else, do just that, using dictionaries:

~~~
{SW1: 5}
~~~

In most cases, nothing like that as needed, as the Pin object is the best and the most efficient identifier of itself:

~~~
if some_pin is SW1:
    print("Switch 1 was pressed")
elif some_pin is SW2:
    print("Switch 2 was pressed")
~~~

closed 2017-05-30 by pfalcon