micropython/micropython #3094: Incorrect documentation for ESP8266 - UART still has the limitations removed from documentation

alexiade, 2017-05-16:
With commit a8ece0358fd5821dd0e9f215f8d852a62de834c0, the documentation was changed to not suppress some methods that are not available on ESP8266, however, they still arent really available..

`
dir(uart)
['init', 'read', 'readline', 'readinto', 'write']
`

alexiade, 2017-05-16:
This limitation is a bit of a pain, as my project needs to check and act on uart - and not USB, but an actual serial device, that occasionally transmits and also intermittently receive commands over MQTT bus... Having no way to do a non-blocking UART read is a show-stopper, so false hope the documentation gives is a bit heartbreaking :P

pfalcon, 2017-05-16:
Right, http://docs.micropython.org/en/latest/esp8266/library/index.html , unmissable colored box with "Warning" title at the top.

As for esp8266, these methods will be missing, until someone who needs them will implement them - properly (which isn't an easy task).

alexiade, 2017-05-16:
I'm aware of the complications. Ive read all of the threads out there about this, plus some source code. What confused me that previously existing supression was removed and and actually about 3 months ago any was added to machine_uart.c and uart.c for the ESP, but my grasp is a bit short as to by magic this would show up in the python.... Apparently it does not. However, timeout now works as confirmed with experimentation. So I can do UART reads with a timeout and that is solving my immediate issue... I just wish the documentation wasn't keeping me chasing ghosts most of the night :)

pfalcon, 2017-05-16:
It wouldn't, if you read it in full. At least you know how to find the ultimate truth for a particular system - by checking the actual module contents ;-). Glad to hear you found a workaround for now.

closed 2017-05-16 by pfalcon

alexiade, 2017-05-16:
:dancer: :P

dpgeorge, 2017-05-17:
ESP8266 UART does have all the code in place for uart.any().  In fact it already works with select/poll, eg to wait up to 2 seconds for a character to be available for reading do:
```python
>>> import select, machine
>>> select.select([machine.UART(0)], [], [], 2)
```
Use select.poll() for more powerful behaviour.

If uart.any() is part of the HW API then we should add it to the esp8266 implementation, and it's easy to do.

pfalcon, 2017-05-17:
> If uart.any() is part of the HW API then we should add it to the esp8266 implementation, and it's easy to do.

Let's better remove it, and use uselect.poll() consistently ;-).

dpgeorge, 2017-05-17:
> Let's better remove it, and use uselect.poll() consistently

Sure, we can, but uart.any() is very convenient (and memory efficient) for quick checking; compare `if uart.any():` with `if select.select([uart], [], [])[0]:`

pfalcon, 2017-05-17:
Then we should decide if it's part of HW API or not ;-). I cast my vote. Btw, select.select is itself optional, only select.poll() is mandatory. If .any() is left there, would be nice to have snippets in docs how to replace it with select.poll().

dpgeorge, 2017-05-17:
If select is optional (I agree with this, since poll is a superset of functionality, and accepts milliseconds as its timeout arg) then we really should keep uart.any() because the alternative using poll is necessarily multi-statement:
```python
poll = select.poll()
poll.register(uart, select.POLLIN)
uart_any = bool(poll.poll(0))
```

alexiade, 2017-05-17:
I'll just say that you guys rock :D

pfalcon, 2017-05-17:
> If select is optional

Unix port doesn't implement it.

>  then we really should keep uart.any()

So, we can reopen this ticket and update title to remind to implement it for esp8266.

alexiade, 2017-05-17:
Just curious for self-education, where does this binding of c to python live in the source tree?

dpgeorge, 2017-05-17:
See #3098 for PR of implementation of uart.any().

dhylands, 2017-05-17:
It happens in modules and types. For example, if you look in modmachine.c:
https://github.com/micropython/micropython/blob/e4a53570d08643d62ba8e2b72919ae1693162e9a/esp8266/modmachine.c#L232

The esp8266 uart is defined by the type pyb_uart_type. If you search for that, you'll find it in machine_uart,c: https://github.com/micropython/micropython/blob/e4a53570d08643d62ba8e2b72919ae1693162e9a/esp8266/machine_uart.c#L283

Most of the methods exposed by a type will be found in the locals dict.

alexiade, 2017-05-17:
Thank you so much for the education And the updates :)

dpgeorge, 2017-05-19:
uart.any() was implemented for esp8266's UART; see aca898eeb0d85a8efbea9abea79c420118521ee9