micropython/micropython #3095: Doc: time.ticks_diff() is not described correctly

riegaz, 2017-05-16:
Old and new is mixed here:

http://docs.micropython.org/en/latest/wipy/library/time.html#time.ticks_diff

pfalcon, 2017-05-16:
@dpgeorge, please clean up dir for generated docs, page above describes itself as v.1.7.

dpgeorge, 2017-05-17:
Ok, stale content was removed, and the docs build script that generates these pages was updated to always remove the old content before generating the new.

@riegaz where did you get that link from?  That link is no longer valid and it might be that there are some stale links out there as well.

riegaz, 2017-05-17:
@dpgeorge It is the first result from Google when searching for "micropython time"

dpgeorge, 2017-05-17:
Ok, thanks, hopefully their cache is updated soon.

closed 2017-05-17 by dpgeorge