micropython/micropython #3097: machine_i2c.c I2C instances don't have correct type

peterhinch, 2017-05-17:
Using machine:
```python
MicroPython v1.8.7-659-g5e66f2b on 2017-05-13; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> from machine import I2C
>>> a = I2C('X')
>>> isinstance(a, I2C)
False
>>> type(a)
<class 'I2C'>
>>> type(a) is I2C
False
>>> 
```
By contrast the pyb module behaves as expected:
```python
>>> from pyb import I2C
>>> a = I2C('X')
>>> isinstance(a, I2C)
True
>>> type(a)
<class 'I2C'>
>>> type(a) is I2C
True
>>> 
```

dpgeorge, 2017-05-17:
That's because there are two I2C types: SoftI2C and HardI2C (which are both rendered as `<class 'I2C'>`).  If your code expects/accepts only either soft or hard I2C peripherals then you can extract and save the type you need, something like:
```python
>>> hard_i2c = type(a)
>>> isinstance(a, hard_i2c)
```
If your code can accept either then you'll need to use hasattr, eg: `hasattr(a, 'readfrom')`.  This is because there's no common base class for the soft and hard classes.

peterhinch, 2017-05-17:
OK, thanks for that.

closed 2017-05-17 by peterhinch