micropython/micropython #3099: mpy-cross makefile error messages 'invalid `override' directive'

peterhinch, 2017-05-18:
Running make clean on mpy-cross and a subsequent build produces the following output. The build is otherwise successful.
```
$ cd mpy-cross
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/mpy-cross
$ make clean
Makefile:4: invalid `override' directive
Makefile:5: invalid `override' directive
Makefile:6: invalid `override' directive
Makefile:7: invalid `override' directive
Makefile:8: invalid `override' directive
Makefile:9: invalid `override' directive
Makefile:10: invalid `override' directive
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -f mpy-cross
rm -f mpy-cross.map
rm -rf build 
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/mpy-cross
$ make -j 8
Makefile:4: invalid `override' directive
Makefile:5: invalid `override' directive
Makefile:6: invalid `override' directive
Makefile:7: invalid `override' directive
Makefile:8: invalid `override' directive
Makefile:9: invalid `override' directive
Makefile:10: invalid `override' directive
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
make: Warning: File `build' has modification time 20 s in the future
mkdir -p build/py
mkdir -p build/py/../extmod
mkdir -p build/py/../lib/embed
mkdir -p build/py/../lib/utils
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
CC ../py/mpstate.c
CC ../py/nlrx86.c
...
```

dpgeorge, 2017-05-18:
I don't get these messages, neither for "make clean" nor "make -j 4".  My make version is:
```
$ make -v
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
...
```

stinos, 2017-05-18:
The Travis build has these messages as well.

peterhinch, 2017-05-18:
Ah. GNU Make 3.81here. I'll upgrade.

dhylands, 2017-05-18:
I have make 3.81 and don't see those messages.

I don't see any override directives in the Makefile either. Perhaps compare your Makefile with the one that's checked in?

peterhinch, 2017-05-19:
See top of [mpy-cross Makefile](https://github.com/micropython/micropython/blob/master/mpy-cross/Makefile).

dhylands, 2017-05-19:
> See top of mpy-cross Makefile.

My bad. My tree was older than the changes. I'm seeing the warning now as well.

dpgeorge, 2017-05-23:
Perhaps the thing to do here is guard those "override undefine" statements by an if which checks that the make version is greater than 3.81?  This will still be a bit of a hack and hopefully replaced soon by somthing better.  Alternatively it can be left as-is and users can upgrade to make v4 if they don't want to see the warnings (it looks like they are just warnings).

PR #3109 (2017-05-26): Fix mpy cross make undefine
Fixes #3099
closed 2017-05-26 by dpgeorge