micropython/micropython #3101: uctypes.sizeof should have explicit layout_type argument

jpochyla, 2017-05-19:
Right now it's impossible to create a buffer with exact size in case of structs with layout different than `LAYOUT_NATIVE`. Example code:

```python
sz = uctypes.sizeof(desc)
buf = bytearray(sz)  # buf gets a length of a struct with added padding, which is too big
struct = uctypes.struct(uctypes.addressof(buf), desc, uctypes.BIG_ENDIAN)
```

If it looks sensible, i'll send a PR right away.

pfalcon, 2017-05-31:
The problem is well-spotted and needs fixing. But please first come up with a proposal of new API for this.

prusnak, 2018-01-01:
Seems this one was forgotten. We use the following patch for quite some time, which adds extra parameter to sizeof, which allows to pass the layout:

https://github.com/trezor/micropython/commit/6c5f953c797ff32911901215b8d0b7615d84ece8

pfalcon, 2018-10-24:
This was implemented with a somewhat different signature, better fitting the intended API of the module in 2411f42ccba561affc0882f8983811b9d8c02b94.

dpgeorge, 2018-10-30:
Closing because this was implemented.

closed 2018-10-30 by dpgeorge