micropython/micropython #3105: Create `map_iter_next()` from core of `dict_iter_next()`, make use of both where appropriate

tomlogic, 2017-05-23:
Working on my first method that processes keyword args, and ran across `dict_iter_next()` as a nice helper for running through the map in a dict.

I see an opportunity to generalize that for any map, and then use the map and dict versions in multiple locations that loop through the map's table and make use of `MP_MAP_SLOT_IS_FILLED()`.

I'm willing to put some effort into making those changes, which should result in reduced overall code size, but want to confirm that there's interest in receiving such a pull request, and confirmation that `map_iter_next()` could live inside of `py/map.c`.

tomlogic, 2017-05-23:
Or just change `dict_iter_next(dict, cur)` to a macro for `map_iter_next(&dict->map, cur)`.

jonnor, 2024-09-19:
Is this still relevant? The linked PR was closed due to too many conflics. CC @dpgeorge