micropython/micropython #3107: Implementation of non-blocking I/O for modussl_mbedtls?

MrSurly, 2017-05-24:
I'm having this problem: [umqtt.simple using ssl: check_msg() throws NotImplementedError](https://github.com/micropython/micropython-lib/issues/168), which led to discovering that mbed_tls [supports non-blocking](https://tls.mbed.org/discussions/feature-request/non-blocking-calls-to-transport-layer-i-o), at least for the version included with the ESP32 IDF.

The issue seems to be that the `ussl` module is using the wrapped socket as a streaming object.  This doesn't allow the proper capture of the `EWOULDBLOCK` signal necessary to generate the `MBEDTLS_ERR_SSL_WANT_READ` and `MBEDTLS_ERR_SSL_WANT_WRITE` codes required by mbed_tls.

Proposed:
1. Extending the `stream` protocol to support timeout / blocking functions -- though this doesn't make much sense, frankly.
2. Creating a new 'socket' protocol that could be sensed by the `ussl` module, thus allowing a call to `setblocking` or `settimeout` to work correctly.  

Are the `ussl` modules intended to wrap things other than sockets?  Could they be more closely coupled if the answer is "no"?

dpgeorge, 2017-05-28:
The axtls implementation of ussl currently only supports blocking mode, and see #2631 for a PR that (attempts to) add non-blocking support.  You may find some ideas in that PR.

MrSurly, 2017-06-01:
My focus is for the ESP32, which uses mbedtls. mbedtls _does_ support blocking. 

I have this working (and using in real code), but it relies on https://github.com/micropython/micropython-esp32/pull/105.

pfalcon, 2017-08-14:
Can this be closed now? Were tests added for the implemented functionality?

MrSurly, 2017-08-14:
>Can this be closed now?

Yes, Closing.

>Were tests added for the implemented functionality?

No.

closed 2017-08-14 by MrSurly