micropython/micropython #3112: statvfs() causes OSError: [Errno 1] EPERM

diginfo, 2017-05-26:
This is fine in 1.87 but won't run in 1.9, has something changed ??

    import os
    os.statvfs('/')

dpgeorge, 2017-05-26:
Lots has changed with the filesystem going from v1.8.7 to v1.9.  But most changes should be invisible to the user.  What port/board/MCU are you using and how is the filesystem configured?  Why do you expect to be able to statvfs the root dir?

diginfo, 2017-05-26:
Sorrie, I am using esp8266

This function gives me the available flash space:

      root = sys.path[1].split('lib')[0]
      s = os.statvfs(root)
      print((s[0]*s[3]) / 1048576,'MB')

dpgeorge, 2017-05-26:
Ok, it looks like this is a bug, that statvfs of the root does not work if something is mounted at root.  To work around it just pass in any string beginning with "/" to statvfs, eg statvfs("//") or statvfs("/lib") (or even a non-existent entry would work, like statvfs("/NOEXIST")).

diginfo, 2017-05-26:
OK, but shouldn't the /flash folder be created automatically and this becomes the root ?

I just re-flashed again and all files including boot.py are still in the root ??

	MicroPython v1.9 on 2017-05-26; ESP module with ESP8266
	Type "help()" for more information.
	>>> import sys
	>>> sys.path
	['', '/lib', '/']
	>>> import os
	>>> os.listdir()
	['boot.py']
	>>>

dpgeorge, 2017-05-27:
> OK, but shouldn't the /flash folder be created automatically and this becomes the root ?

On the esp8266 "/" is the root and is where the internal filesystem is mounted; there is no "/flash" (it just worked for you above in statvfs because *any* path will work).  See #2920 and #2950 for previous discussion about this topic.

pacmac, 2017-05-27:
OK, no problem that's how it has always been, but I seem to recall reading somewhere that there were plans to standardise the /flash folder across all devices, but I don't recall where that was.

dpgeorge, 2017-06-07:
This was fixed in f6ef8e3f17222a397e02f93a8b0d283e0f6c9793

closed 2017-06-07 by dpgeorge